microsoft/AirSim #1734: Human Ground truth locations

MahirGulzar, 2019-01-24:
Hi is it possible to extract humanoid models location relative to camera while recording? Currently the airsim_rec text file gives information related timestamps and frames. What about extracting other data as well? Not only human locations may be some other objects in world space as ground truth?

PS: I am using Unreal 4.21 on windows 10.

madratman, 2019-02-19:
You'll have to add that feature by using get actor location and get static mesh locations using unreal engine apis. 

Look at how we're the drone and camera positions here

closed 2019-02-19 by madratman

madratman, 2019-02-19:
also, we have [simGetObjectPose](https://github.com/Microsoft/AirSim/blob/master/docs/apis.md#common-apis)

relevant issues https://github.com/Microsoft/AirSim/issues/1332 and https://github.com/Microsoft/AirSim/issues/1014