microsoft/AirSim #1740: Unreal Vehicle Content in CityEnviron

DTChuck, 2019-01-25:
I am working towards creating a customized environment that is like the CityEnviron environment, but with some specific requirements about the vehicles. Is there any information on the asset packs and vehicle/pedestrian AI that I can access to expedite my environment creation?

Using Unreal 4.18, Windows 10.

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

msb336, 2019-01-29:
@RVBldr is there documentation about which packages are getting used for different environments?

jonyMarino, 2021-08-19:
See comment on the latest release please

closed 2021-08-19 by jonyMarino