microsoft/AirSim #1742: Issue with MultirotorRpcLibServer

Vinay545, 2019-01-26:
I am working Airsim in Ubuntu Linux.. I am using the environments i.e. binaries. I have done some changes in "MultirotorRpcLibServer.cpp" of AirSim plugin library. As the library is static, I cannot execute those changes while running simulator. Is there any way to solve this problem. 

One solution is running the simulator in windows, where we can build the environment. But I want to run in ubuntu.

msb336, 2019-01-29:
@Vinay545 
You can rebuild the airsim library to incorporate your changes by running the `./build.sh` command inside the main directory.

madratman, 2019-02-19:
@Vinay545 umm, no. @msb336 is correct, just rebuild the plugin with `./build.sh`. 
it will indeed change server part. 

you don't need to rebuild unreal engine (unreal engine's buildtool is a bit wonky, especially in linux, so if it recompiles itself when you change the problem, we can't really do anything about it apart from blaming unreal)

to port the new plugin to a new environment, you jsut need to copy the airlib plugin to the new project as explained in teh dev-workflow article - https://microsoft.github.io/AirSim/docs/dev_workflow/

closed 2019-02-19 by madratman