microsoft/AirSim #1744: Using Multiple cameras for detection

Ahmad5112, 2019-01-27:
I want to use multiple cameras for capturing the image and based on the depth image being captured by the cameras I want to make an array of distances and based on that array I want to run a learning algorithm .is it possible to do it so how can I do it

AloshkaD, 2019-01-28:
Yes, it is possible but your question is really broad. Try to start with a simple example, then implement it. Break down what you need to accomplish into algorithms and build the architecture then build the components and put them together. What you are asking to do is a big project that consists of too many components and no one can really show you how to do everything like that.

Ahmad5112, 2019-01-28:
yes you are right but in the meantime, my question is how can I implement multiple cameras and based on distance how can i make Array of distance using depth image

madratman, 2019-02-19:
@Ahmad5112, please take @AloshkaD's comment seriously. Please keep your issues to the point.

closed 2021-11-30 by jonyMarino