microsoft/AirSim #1753: Airsim Build error: Cmake Windows

ethbond, 2019-01-30:
Attempting to build Airsim, error reading what version cmake I have.
did cmake --version to check installed version (see screen cap), seems to be fine any ideas? 

[
[image]
](url)

samuelgarcia321, 2019-01-30:
try install module C++ In Microsoft Visual Studio.
its worked for me.

arian-amador, 2019-01-30:
Initially looking at this it seems like an issue with `find` or `cmake` but then noticed that you cloned the AirSim repo into your `c:\Windows\System32` directory AND you're also running the Native Tools CLI as an admin?

I recommend you try and start the native tools as a normal user and clone the AirSim repo into another non-system directory. Example: `c:\AirSim\`

ethbond, 2019-01-31:
@samuelgarcia321 So I verified that the C++ module was installed (see images) also included a list of the individual components perhaps I have missed something?
[image]
[image]

@arian-amador Moved it to my D: drive, ran as a normal user rather than admin, same problem.
[image]

madratman, 2019-02-19:
@ethbond, were you able to compile it? 
can you try reinstalling cmake >=3.9 from here https://cmake.org/download/

ethbond, 2019-02-19:
I managed to compile it, but by doing a workaround. The cmake install is fine. The problem seems to be with Check_cmake.bat, failing to find the cmake version, or writing the version it finds to the "cmake_version" variable. My workaround was to just type in my version into line 8 of Check_cmake.bat:

 set "cmake_version=    3.     13"

Saved the .bat, ran the build again and it worked fine.

madratman, 2019-02-19:
@ethbond thanks for getting back quickly 
hmm. that's weird, I haven't seen any other issue in which cmake wasn't found..
ok, I did a quick hunt again in old issues, and this comment seems related https://github.com/Microsoft/AirSim/issues/755#issuecomment-423298213

The issue seems to be this line https://github.com/Microsoft/AirSim/blob/master/check_cmake.bat#L12
`'cmake --version ^| find "cmake version"'`

Can you please share the output of `where Find.exe` on your machine? Maybe you and @vincedjango had similar causes. 

In my machine, I don't have cygwin, so it's simply:
```
C:\Users\ramadaan>where Find.exe
C:\Windows\System32\find.exe
```

closed 2020-09-04 by jonyMarino

ibadrather, 2022-08-23:
> @madratman
> 
> I managed to compile it, but by doing a workaround. The cmake install is fine. The problem seems to be with Check_cmake.bat, failing to find the cmake version, or writing the version it finds to the "cmake_version" variable. My workaround was to just type in my version into line 8 of Check_cmake.bat:
> 
> set "cmake_version= 3. 13"
> 
> Saved the .bat, ran the build again and it worked fine.

Thank you for your solution. Worked for me too.