microsoft/AirSim #1754: Car in city environment driving in circles

NextSim, 2019-01-30:
I just downloaded the release city environment 1.2.1, extracted it and run run.bat.

The city loads, and the car starts driving with an accel of -0.5, steering of 1.5 and it is in reverse.

I don't have any python scripts running, so it isn't like I am controlling it over the api.  It says control mode is keyboard.

If I press the up key it will start driving forward.  If I press the left arrow key the steering decreases by 1 to 0.5, pressing the right key increases the steering to 2.5.

msb336, 2019-01-30:
@NextSim is this in the windows, or linux environment?

NextSim, 2019-01-31:
Windows 7, 64bit

airsimbug, 2019-07-09:
Is anyone working on this? It's happening to me on Windows 10, 64bit.

airsimbug, 2019-07-09:
This is frustrating! Why is this happening to me? I'm on Windows 10, 64bit.

airsimbug, 2019-07-09:
This stopped happening after I disabled a Virtual Hid Device in Device Manager > Human Interface Device. I had a Virtual Hid Device in Game Controllers.

closed 2020-02-14 by ironclownfish