microsoft/AirSim #1755: python simGetImages API bug

flyers, 2019-01-30:
I use the following code to get the segmentation image from the camera.

```python
responses = client.simGetImages([ImageRequest("0", airsim.ImageType.Segmentation, False, False)])
response = responses[0]

# get numpy array
img1d = np.fromstring(response.image_data_uint8, dtype=np.uint8) 

# reshape array to 4 channel image array H X W X 4
img_rgba = img1d.reshape(response.height, response.width, 4)  
```

However, the following error happens very randomly.
```
ValueError: cannot reshape array of size 1 into shape (0,0,4)
```
It seems that sometimes the returned response will have zero height and width and the data only have a length of 1.
Similar problems have been reported in #1710.
I believe it's a serious bug in the simGetImages API.

srivatsankrishnan, 2019-02-01:
@flyers @sytelus I am seeing this issue too. But I do have a workaround for this issue. I have a `try` and `except`  before the simGetImages call. On exception, I try to reset AirSim and pass the dummy image. Something like this:
```
try:
   image = client.simGetImage(....)
except Exception as e:
    client.reset()
     image = np.ones(image.shape)
```

I have seen this workaround recovering from that error. But this is definitely a hack and it would be great if this is fixed in AirSim.

ElliotHYLee, 2019-02-02:
Also, after this lines:
responses = client.simGetImages([
        airsim.ImageRequest("0", airsim.ImageType.DepthVis),
        airsim.ImageRequest("1", airsim.ImageType.DepthPerspective, True),
        airsim.ImageRequest("2", airsim.ImageType.Segmentation),
        airsim.ImageRequest("3", airsim.ImageType.Scene),
        airsim.ImageRequest("4", airsim.ImageType.DisparityNormalized),
        airsim.ImageRequest("4", airsim.ImageType.SurfaceNormals)])

for i, response in enumerate(responses):
    if response.image_type == airsim.ImageType.Scene:
        airsim.write_file(str(i) + '.png', response.image_data_uint8)

I have this image

[image]

Instead of this:
[image]

gurkanctn, 2019-02-03:
Until a permanent solution is found, another workaround can be using an `if` statement to check for the length of the response, and only try reshaping if it passes the test. This might be faster than the try-catch (to be confirmed).

annisanazi, 2020-02-13:
did anyone resolve this issue?

rallen10, 2020-07-22:
I am still facing this issue! There are many more issues like this open and the ones that are closed are only closed because someone recommended the `if` statement workaround. Can we get a higher priority on this?