microsoft/AirSim #1756: Python Client Connection for API use

Benji15150, 2019-01-31:
Hello, 

I am sorry I might be asking obvious question, but I am completely new to Airsim and Unreal Engine. I am having trouble using python API, because when I tried to open hello_car, it raises the connection error like below: 
[image]
I am using ubuntu machine, and the airsimgamemore works fine in unreal engine so I can manipulate car with arrow keys when I press play button. What can I do to solve the above connection problem? 
Also, if it connects normally, would running the python command in the shell open unreal engine and show simulation by default? or is there other step that is required?? Thank you for the help in advance!

msb336, 2019-01-31:
@Benji15150
Can you add a comment with your settings.json file to this thread

Benji15150, 2019-01-31:
`{"SettingsVersion":1.2,
  "SimMode": "Car}`

closed 2020-08-17 by jonyMarino