microsoft/AirSim #1758: RPCLib not found while running build.cmd

suoyita, 2019-02-01:
I am trying to build airsim on Windows 10 with Visual Studio 2017 and Unreal 4.18

The message in cmd is as follows:

================================================

C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\AirSim>build.cmd
Found cmake version: 3.13.3
The directory or file cannot be created.

*****************************************************************************************
Downloading rpclib
*****************************************************************************************
iwr : Could not find a part of the path 'C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\AirSim\external\rpclib.zip'.
At line:1 char:86
+ ... pe]::Tls12; iwr https://github.com/rpclib/rpclib/archive/v2.2.1.zip - ...
+                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [Invoke-WebRequest], DirectoryNotFoundException
    + FullyQualifiedErrorId : System.IO.DirectoryNotFoundException,Microsoft.PowerShell.Commands.InvokeWebRequestCommand

The system cannot find the path specified.
New-Item : Access to the path 'rpclib' is denied.
At C:\WINDOWS\system32\WindowsPowerShell\v1.0\Modules\Microsoft.PowerShell.Archive\Microsoft.PowerShell.Archive.psm1:318 char:32
+ ... eatedItem = New-Item -Path $DestinationPath -ItemType Directory -Conf ...
+                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : PermissionDenied: (C:\Program File...external\rpclib:String) [New-Item], UnauthorizedAccessException
    + FullyQualifiedErrorId : CreateDirectoryUnauthorizedAccessError,Microsoft.PowerShell.Commands.NewItemCommand

The system cannot find the file specified.
Unable to download high-polycount SUV. Your AirSim build will use the default vehicle.

#### Build failed - see messages above.

==============================================

madratman, 2019-02-19:
This is weird. My guess is this a permissions issue coz of ` New-Item : Access to the path 'rpclib' is denied.`, as you're in `C:\Program Files (x86)\`. You can try cloning airsim repo in something like `C:\Users\USERNAME\...` and try again. 
 
`build.cmd` should download rpclib for you, as is evident in Lines 48 to 74 here https://github.com/Microsoft/AirSim/blob/master/build.cmd#L48-#L74
You can extract out these lines run them manually in the command line, and see if you're able to get rpc

closed 2019-02-19 by madratman

MHSHAN925, 2022-02-05:
do you have soved this problem?