microsoft/AirSim #1759: Access API in Unreal

thomasvolpato, 2019-02-01:
Hi,
  sorry for my noob questions, but how do I access the classes of the API?
  I wish to access the LIDAR Data using my own class.
  I do not know where should I include the headers and references,
  also, if this is the correct way to use the API

## Steps:
* Created a new C++ Class derived from AActor (AAnalyzer)
* now, if I try to access the VehicleApiBase, it does not work

>#include "Analyzer.h"
>// Called when the game starts or when spawned
>void AActuator::BeginPlay()
>{
>	Super::BeginPlay();
>
>	**msr::airlib::VehicleApiBase** * car = FindComponentByClass<**msr::airlib::VehicleApiBase**>();
>	if(car)
>		UE_LOG(LogTemp, Warning, TEXT("Sensors: %d"), car->getSensors().size)
>}

msb336, 2019-02-04:
@thomasvolpato 
This may be an x y problem. If you are just trying to access the lidar data and modify it, it will be much easier to implement it on the client side.

```
class LidarManipulator()
{
    public:
         msr::airlib::MultirotorRpcLibClient client_;
        msr::airlib::LidarData lidar_;

        void getLidar()
        {
            lidar_ = client_.getLidarData()
            do *stuff*
        }
}

thomasvolpato, 2019-02-05:
ok! Thank you.
 however, how do I link the API with Unreal?

1. I tried to use:

```c++

#include "CoreMinimal.h"
#include "GameFramework/Actor.h"
#include "vehicles/multirotor/api/MultirotorRpcLibClient.hpp" //<--------- added line
#include "Analyzer.generated.h"

UCLASS()
class LIDAREXAMPLE_API AAnalyzer: public AActor
{
	GENERATED_BODY()
public:	
	AAnalyzer();
	msr::airlib::MultirotorRpcLibClient client_; //<--------- added line

protected:
	// Called when the game starts or when spawned
	virtual void BeginPlay() override;

public:	
	// Called every frame
	virtual void Tick(float DeltaTime) override;
};
```

2. and I got the "error cannot open include file ... "
3. so I added the include path in the Project->Properties/VC++ Directories->Include Directories. However, now I have the following errors:

> error LNK2019: unresolved external symbol "public: __cdecl msr::airlib::MultirotorRpcLibClient::MultirotorRpcLibClient( ...

> error LNK2001: unresolved external symbol "public: __cdecl msr::airlib::MultirotorRpcLibClient::MultirotorRpcLibClient( ...

>error LNK2019: unresolved external symbol "public: virtual __cdecl msr::airlib::MultirotorRpcLibClient::~MultirotorRpcLibClient(void)" ...

> error LNK2001: unresolved external symbol "public: virtual __cdecl msr::airlib::MultirotorRpcLibClient::~MultirotorRpcLibClient(void)" ...

> error LNK1120: 2 unresolved externals	LidarExample E:\LidarExample\Binaries\Win64\UE4Editor-LidarExample.dll

> UBT ERROR: Failed to produce item: E:\LidarExample\Binaries\Win64\UE4Editor-LidarExample.dll	LidarExample	E:\UnrealProjects\LidarTest\LidarExample\Intermediate\ProjectFiles\UnrealBuildTool	1	

> error MSB3075: The command ""C:\Program Files\Epic Games\UE_4.21\Engine\Build\BatchFiles\Build.bat" LidarExampleEditor Win64 Development "E:\LidarExample\LidarExample.uproject" -WaitMutex -FromMsBuild" exited with code 5. Please verify that you have sufficient rights to run this command.

thomasvolpato, 2019-02-06:
Sorry @msb336 but my question is how to use the API inside Unreal. If I modify the source inside Unreal/Plugins, it works. But I think this is not the correct way to use a plugin.

msb336, 2019-02-06:
@thomasvolpato 
Are you trying to modify the plugin source code? Otherwise, definitely do not change files inside of the plugin library. If you are asking how to run airsim with an unreal environment, here is a [tutorial](https://microsoft.github.io/AirSim/docs/unreal_blocks/) to get up and running using the blocks environment.

With airsim, two things are happening:
1. You're executable, which you have built to do perform whatever drone-related functions you choose (i.e. hellodrone.exe). This sends commands to the airsim server.
2. The Unreal environment has an airsim plugin which acts as a server, receiving requests from your executable, and converting them into unreal commands

thomasvolpato, 2019-02-07:
I had a issue trying to compile the code above.
I figured out that I had to include the following code in the file
***ProjectName/Source/ProjectName/ProjectName.build.cs***

```c++
PublicIncludePaths.Add("E:/ProjectName/Plugins/AirSim/Source/AirLib/include");
PublicIncludePaths.Add("E:/ProjectName/Plugins/AirSim/Source/AirLib/deps/eigen3");
PublicIncludePaths.Add("E:/ProjectName/Plugins/AirSim/Source/AirLib/deps/MavLinkCom/include");
PublicIncludePaths.Add("E:/ProjectName/Plugins/AirSim/Source/AirLib/deps/rpclib/include");

PublicAdditionalLibraries.Add("E:/ProjectName/Plugins/AirSim/Source/AirLib/lib/x64/Release/AirLib.lib");
PublicAdditionalLibraries.Add("E:/ProjectName/Plugins/AirSim/Source/AirLib/deps/rpclib/lib/x64/Release/rpc.lib");
PublicAdditionalLibraries.Add("E:/ProjectName/Plugins/AirSim/Source/AirLib/deps/MavLinkCom/lib/x64/Release/MavLinkCom.lib");
```

coldwarrior5, 2019-02-08:
Hi,
so I have also tried including everything as stated but I get linker errors. 
I have included all the libraries, includes, and rpc.lib in the new client-side project and I tried copying HelloCar.cpp.
The Intelisense is not complaining, definitions are available, however, upon compilation, I get linker errors claiming that the methods cannot be found. 
I am compiling using Debug setting the x64 platform for the Windows 8.1 target.

hangmann, 2019-07-26:
@coldwarrior5 Could you solve the problem?

coldwarrior5, 2019-07-26:
The solution from thomas the one above my last comment worked for me as well.
You just have to include additional libraries even though they didn't mention it.

denmonz, 2022-01-16:
Just wanted to add to the conversation, as I was also having issues:

If you're still getting the error that the specified '.exe' cannot be found, you need to set the 'Solution Configuration' to 'Debug' and the 'Solution Platforms' to 'x64' before launching the 'Local Windows Debugger' in AirSim.sln. If you're launching the simulation environment from VS as well (e.g., the provided 'Blocks'), be sure that the settings are 'DebugGame' and 'Win64'.

Then, launch your simulation environment (e.g., 'Blocks') and debug HelloDrone.