microsoft/AirSim #1760: CarClient Linear Velocity

ElliotHYLee, 2019-02-02:
From version 1.2.0, for the release versions, the car mode's linear velocity is not [0,0,0]. It seems the velocities differ by the environment package. However, the MultirotorClient's linear velocity are all [0,0,0] at the beginning.

What's causing this? and can there be a fix for this using python api?

dawgster, 2019-02-10:
I'm having the same issue. Is this a bug or a feature I don't understand yet? In the blocks environment, a value for linear_velocity.x_val of around -59 shows when in standstill.

ironclownfish, 2020-05-20:
After PR 2705 is merged, you can safely do this to achieve true pause.
[image]

closed 2020-07-22 by madratman