microsoft/AirSim #1761: Speed of simGetImages()

ElliotHYLee, 2019-02-02:
Using python and release 1.1.9 with neighborhood environment, when I call simGetImages(), it takes about 0.1 seconds. In this way, I can collect data only at ~ 10Hz, can there be a good way to collect the data at fater rate?

FYI, the reason why I am not using the R key with the corresponding setting.json is that I cannot collect velocity information with it. Or, I can change the code in the source and rebuilt it. Yet, I do not have neighborhood like environment, which I need for the consistency.

madratman, 2019-02-19:
This is indeed a performance limitation as of now. Here are three suggestions which might make life better:  
- Get uncompressed images, that's the fastest possible option, as of now. 
I guess you are already doing that, if not, see here https://github.com/Microsoft/AirSim/issues/1786#issuecomment-465228527 for how to use the api
- In your settings.json, you should disable all subwindows for a small performance gain
- You can also try running in the headless mode in ubuntu for a small boost, as explained here https://github.com/Microsoft/AirSim/issues/1781#issuecomment-465234945
- No brainer, but decrease camera resolution, if feasible. 

There are some more ideas in this issue https://github.com/Microsoft/AirSim/issues/796. Tagging @sytelus if he has any more ideas.

closed 2019-02-19 by madratman

elmtalab, 2019-06-18:
you can collect images every 0.003 second with screen capturing. you may lost your screen for a while but you can collect your training data for a night and work on it afterwards. see the work of @Frannecklp. you can also find data set collection codes [here](https://github.com/Sentdex/pygta5/blob/master/1.%20collect_data.py).