microsoft/AirSim #1762: Camera Capture Default Settings

Gaduu, 2019-02-03:
Quick question regarding AirSim default settings. In the example below the settings are applied to the Scene ImageType, my aim is to change width and height not to only the scene image type but also to segmentation and depth perspective, is there any way to do it? I've tried adding another line e.g. "ImageType:5" below the first one (ImageType:0) but it has overridden the scene settings i.e. segmentation width and height was indeed 1080x720 but the scene image size was set back to the default size i.e. ~200x~144. Appreciate any help.

"CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 1080,
        "Height": 720
      }

Cheers,

Gaduu, 2019-02-04:
Closed. Figured out the answer to my question. If anyone is looking for the solution to this it is just matter of adding another block of: 
{
"ImageType": n,
"Width": 1080,
"Height": 720
}

in a single CaputreSettings[] block

closed 2019-02-04 by Gaduu

majid1268, 2019-04-13:
Could you please share your settings.json file? 
I am also trying to change the default values from 255x144 into 640x448 (Width and height) but I was not successful. I changed the settings.json file based on the doc in [https://github.com/Microsoft/AirSim/blob/master/docs/settings.md](URL). But unfortunately, the values are still unchanged.
My setting file is: 

`{
"SettingsVersion": 1.2,
"CameraDefaults": {
"CaptureSettings": [
{
"ImageType": 0,
"Width": 640,
"Height": 448,
"FOV_Degrees": 70,
"AutoExposureSpeed": 100,
"MotionBlurAmount": 0
},
{
"ImageType": 2,
"Width": 640,
"Height": 448,
"FOV_Degrees": 70,
"AutoExposureSpeed": 100,
"MotionBlurAmount": 0
}
]
},
"PhysXCar": {
      "VehicleType": "PhysXCar",
      "Cameras": {"CaptureSettings": [
      {
      "ImageType": 0,
      "Width": 640,
      "Height": 448,
      "FOV_Degrees": 70,
      "AutoExposureSpeed": 100,
      "MotionBlurAmount": 0
      },
      {
      "ImageType": 2,
      "Width": 640,
      "Height": 448,
      "FOV_Degrees": 70,
      "AutoExposureSpeed": 100,
      "MotionBlurAmount": 0
      }
      ]
      }
    },
"SimMode": ""
}`

I will appreciate if you can help me to understand where is my problem.