microsoft/AirSim #1766: Calling simGetImage or simGetImages once drops performance permanently. Why?

ypvypv, 2019-02-04:
Calling simGetImage or simGetImages from python or C++ permanently impacts rendering framerate. In my case I am running at 60fps generally and can communicate with unreal via other airsim lightweight APIs without framerate costs.

As soon as simGetImage or simGetImages is called, the framerate drops to 20fps on my machine. Even if I call simGetImage once, the framerate is reduced. Can you explain where in the code the permanent cost is paid? You must be scheduling simGetImage tasks continuously in Unreal as RenderTasks, are you? Why does the performance drop permanently?

The recording button ( because it calls the same api ) will also drop performance at the same rate.

I understand that you optimized the performance ( ReadSurfaceData and not ReadPixels ) to bring it up to the current level, as you mention here:
https://stackoverflow.com/questions/43218678/unreal-engine-accessing-saving-in-game-images-to-disk-without-blocking-game-thr/43287505

Also interesting to mention that the actual performance drop is the same whether you are capturing several camera views or just one, which makes me feel that the cost is paid out of render thread management.

NextSim, 2019-02-04:
Following... trying to capture data for machine learning and this is killing e

marcociara379, 2019-02-22:
following , it affects me as well

NextSim, 2019-02-22:
anyone from Microsoft who can respond? even at least to answer @ypvypv 's questions?

NextSim, 2019-03-26:
@sytelus   any thoughts / comments on this?  I can't see how anyone is doing machine learning or autonomous driving with this issue

ironclownfish, 2020-05-20:
You will be able to de-activate the cameras to speed things back up after PR 2465 is merged.

closed 2020-09-02 by madratman