microsoft/AirSim #1771: AirSim + PX4 SITL cannot fly properly

marcociara379, 2019-02-06:
Hi everyone, 
I am currently having problems with PX4 (checkout at v1.8.2, linux on windows subsystem) SITL in AirSim (checkout master) on Windows 10.
I am modifying the quadcopter mass and thrust to mimic mine, in particular the following lines:
- ` real_T max_thrust = 20.179446268f;`  
-  `real_T max_torque = 1.055562f; //computed from above formula`  
- `real_T max_rpm = 7200.0f;`  
- commenting `max_thrust = C_T * air_density * nsquared * pow(propeller_diameter, 4);`  
- commenting `max_torque = C_P * air_density * nsquared * pow(propeller_diameter, 5) / (2 * M_PIf);`  
- `params.mass = 4.0f;` in file `AirSim\AirLib\include\vehicles\multirotor\firmwares\mavlink`  

After building AirSim (no errors) and opening AirSim with PX4 in SITL,, when I "commander arm" the PX4 SITL output:  
`ERROR [commander] CRITICAL NAVIGATION FAILURE - CHECK SENSOR CALIBRATION`
and `WARN  [ecl/EKF] EKF baro hgt timeout - reset to baro`. The quadcopter in the simulation goes "crazy", taking off just with the arm command and resulting in total control loss. I have tried to put to zero all the PID rate gains to check if it was a control issue, but the behavior is the same.

Final note: everything works perfectly with default thrust and mass of the Quadcopter.

Do you have an idea of why there is such a behavior?

Thanks

lovettchris, 2019-04-10:
Note: I found a problem with PX4 1.8.2, they removed a module that is needed for HITL mode to work, namely, "pwm_sim_out" and so I built a new version of the v1.8.2 with this module included and takeoff and normally flying works again. See [OneDrive download](https://1drv.ms/f/s!AoS-ymEfLuCLqudxN4NCgYtf0f3WPQ).  This probably won't fix the above issues, this is just an FYI.

jonyMarino, 2021-06-03:
Everything working on the latest version.

closed 2021-06-03 by jonyMarino