microsoft/AirSim #1773: Question: Get Car Controls

Gaduu, 2019-02-06:
Is there any way to retrieve car's current controls e.g. steering, brake, acceleration (throttle)?
Also, noticed that getCarState().handbrake for some reason always returns False even though the documentation says that the handbrake is active when Space is pressed.

msb336, 2019-02-09:
@Gaduu Handbrake has now been bound to the `end` key, and is a separate property than footbrake, we'll update to docs to reflect this.

As far as getting the car controls, there is a function inside of the airsim plugin that allows you to do this, but it is not bound to an rpc message, so it can't be called by the airsim client. Thanks for bringing that to my attention, I'll rectify that in a PR

Gaduu, 2019-02-11:
Thanks a lot, for now I could only recommend to others to use keyboard.is_pressed('button') to check if you for instance press a key that corresponds to a specific action. E.g. knowing that arrow up corresponds to setting throttle = 1 you can just check if the key is pressed and sort of hard code this value. 

Cheers,

closed 2019-02-11 by Gaduu