microsoft/AirSim #1774: Problem with DQNcar.py

willydra, 2019-02-06:
Hi guys,
Sorry for my noob question.
I tried to run DQNcar.py, after the Airsim App opened.
From the observation, I still don't know that why the agent is always done and reset the position without changing any position, although there is a change of throttle value become 1, but when I stop the program, the car throttle value is increased and the car just run by itself. It looks like the agent just received car_control after the program is stopped. There is no error in console. I don't know for sure, what is the cause of this problem. I attach the video of my trial.

https://www.youtube.com/watch?v=kxyMXjFn9Jw&feature=youtu.be

Thank you very much for your help.

msb336, 2019-02-09:
@willydra the link you have attached is broken.
However, I've seen this behavior before. if you look at the source code for DQNcar.py, you can see the following terminal conditions:
https://github.com/Microsoft/AirSim/blob/9e919a8f69d498243b09cc96a7ba74fa38c48abc/PythonClient/car/DQNcar.py#L486-L493

Which get pinged one second after a reset:
https://github.com/Microsoft/AirSim/blob/9e919a8f69d498243b09cc96a7ba74fa38c48abc/PythonClient/car/DQNcar.py#L528-L533

Depending on how powerful your PC is, it might not be able to render enough frames in unreal to get the car moving, resulting in the time going by too quickly. To compensate, you can increase the sleep time in line 532.

It drives off into the sunset after a reset because default control is throttle=1. This is the last command sent before closing, so there is no command telling the car to stop.

msb336, 2019-02-19:
Closing this thread due to inactivity. If your issue is not resolved, please re-open the thread

closed 2019-02-19 by msb336

MoritzGa, 2019-05-29:
I have the same issue but I'm running in this on a machine with 16 Cores, Titan V and 64 GB of RAM. Also the GPU isn't even used to its full capacity.
In my opinion the issue is that the car will be reset instantaneously as the velocity is below the minimum velocity.

DAndrik21, 2021-12-15:
> I have the same issue but I'm running in this on a machine with 16 Cores, Titan V and 64 GB of RAM. Also the GPU isn't even used to its full capacity. In my opinion the issue is that the car will be reset instantaneously as the velocity is below the minimum velocity.

tried this with min_speed 0 , still having an issue