microsoft/AirSim #1778: Error running Setup.sh on Ubuntu

Benji15150, 2019-02-07:
Hello, I am attempting to build and setup Airsim. All I have done for now is cloned AirSim and ran ./setup.sh. The following is the full message I get. But when I try to run ./build.sh its giving me error that llvm-source-50 folder was not found. Any suggestions to solve the problem?: 
+ set -e
+++ dirname ./setup.sh
++ cd .
++ pwd
+ SCRIPT_DIR=/home/biorobotics/airsim_new/AirSim
+ pushd /home/biorobotics/airsim_new/AirSim
+ downloadHighPolySuv=true
+ [[ '' == \-\-\n\o\-\f\u\l\l\-\p\o\l\y\-\c\a\r ]]
++ uname
+ '[' Linux == Darwin ']'
+ [[ ! -z '' ]]
+ sudo apt-get install -y build-essential
[sudo] password for biorobotics: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.1ubuntu2).
The following packages were automatically installed and are no longer required:
  fltk1.3-doc fluid gir1.2-gtk-2.0 hddtemp libatk1.0-dev
  libcairo-script-interpreter2 libcairo2-dev libclang1-3.6
  libcollada-dom2.4-dp-dev libcollada-dom2.4-dp0 libfltk-cairo1.3
  libfltk-forms1.3 libfltk-gl1.3 libfltk-images1.3 libfltk1.3 libfltk1.3-dev
  libgdk-pixbuf2.0-dev libgtk2.0-dev libharfbuzz-dev libharfbuzz-gobject0
  libjs-jquery-ui libllvm3.6v5 libpango1.0-dev libpixman-1-dev
  liburdfdom-tools libwxsqlite3-3.0-0 libxcb-shm0-dev libxcomposite-dev
  libxcursor-dev libxinerama-dev libxml2-utils libxrandr-dev
  linux-headers-4.15.0-36 linux-headers-4.15.0-36-generic
  linux-headers-4.15.0-42 linux-headers-4.15.0-42-generic
  linux-image-4.15.0-36-generic linux-image-4.15.0-42-generic
  linux-modules-4.15.0-36-generic linux-modules-4.15.0-42-generic
  linux-modules-extra-4.15.0-36-generic linux-modules-extra-4.15.0-42-generic
  python-cycler python-matplotlib python-matplotlib-data python-opengl
  python-psutil python-pydot python-pyqt5.qtopengl python-pyqt5.qtwebkit
  python-tz ros-kinetic-actionlib-tutorials ros-kinetic-bond-core
  ros-kinetic-camera-calibration ros-kinetic-carrot-planner
  ros-kinetic-collada-parser ros-kinetic-collada-urdf ros-kinetic-common-msgs
  ros-kinetic-common-tutorials ros-kinetic-compressed-depth-image-transport
  ros-kinetic-desktop ros-kinetic-diagnostic-analysis
  ros-kinetic-diagnostic-common-diagnostics ros-kinetic-diagnostics
  ros-kinetic-eigen-stl-containers ros-kinetic-executive-smach
  ros-kinetic-fake-localization ros-kinetic-filters ros-kinetic-gazebo-msgs
  ros-kinetic-geometric-shapes ros-kinetic-geometry
  ros-kinetic-geometry-tutorials ros-kinetic-gl-dependency
  ros-kinetic-global-planner ros-kinetic-image-common
  ros-kinetic-image-pipeline ros-kinetic-image-publisher
  ros-kinetic-image-rotate ros-kinetic-image-transport-plugins
  ros-kinetic-image-view ros-kinetic-interactive-marker-tutorials
  ros-kinetic-laser-assembler ros-kinetic-laser-filters
  ros-kinetic-laser-pipeline ros-kinetic-librviz-tutorial ros-kinetic-mk
  ros-kinetic-move-slow-and-clear ros-kinetic-navigation
  ros-kinetic-nodelet-core ros-kinetic-nodelet-tutorial-math
  ros-kinetic-octomap ros-kinetic-perception ros-kinetic-perception-pcl
  ros-kinetic-pluginlib-tutorials ros-kinetic-polled-camera
  ros-kinetic-qt-dotgraph ros-kinetic-qt-gui-cpp ros-kinetic-qwt-dependency
  ros-kinetic-random-numbers ros-kinetic-robot ros-kinetic-robot-model
  ros-kinetic-ros ros-kinetic-ros-base ros-kinetic-ros-comm
  ros-kinetic-ros-core ros-kinetic-ros-tutorials ros-kinetic-rosboost-cfg
  ros-kinetic-roscpp-core ros-kinetic-roscpp-tutorials ros-kinetic-roscreate
  ros-kinetic-roslang ros-kinetic-roslint ros-kinetic-roslisp
  ros-kinetic-rosmake ros-kinetic-rospy-message-converter
  ros-kinetic-rqt-action ros-kinetic-rqt-bag-plugins
  ros-kinetic-rqt-common-plugins ros-kinetic-rqt-dep ros-kinetic-rqt-graph
  ros-kinetic-rqt-gui-cpp ros-kinetic-rqt-image-view ros-kinetic-rqt-launch
  ros-kinetic-rqt-moveit ros-kinetic-rqt-msg ros-kinetic-rqt-plot
  ros-kinetic-rqt-pose-view ros-kinetic-rqt-publisher
  ros-kinetic-rqt-py-console ros-kinetic-rqt-reconfigure
  ros-kinetic-rqt-robot-plugins ros-kinetic-rqt-robot-steering
  ros-kinetic-rqt-runtime-monitor ros-kinetic-rqt-rviz
  ros-kinetic-rqt-service-caller ros-kinetic-rqt-shell ros-kinetic-rqt-srv
[... 100 lines cut ...]
Ign:25 http://repos.codelite.org/ubuntu uni/iverse Translation-en_US
Ign:26 http://repos.codelite.org/ubuntu uni/iverse Translation-en              
Ign:27 http://repos.codelite.org/ubuntu uni/iverse amd64 DEP-11 Metadata       
Ign:28 http://repos.codelite.org/ubuntu uni/iverse DEP-11 64x64 Icons          
Fetched 3,429 kB in 6s (536 kB/s)                                              
Reading package lists... Done
W: Target Packages (universe/binary-amd64/Packages) is configured multiple times in /etc/apt/sources.list.d/codelite.list:1 and /etc/apt/sources.list.d/codelite.list:2
W: Target Packages (universe/binary-i386/Packages) is configured multiple times in /etc/apt/sources.list.d/codelite.list:1 and /etc/apt/sources.list.d/codelite.list:2
W: Target Packages (universe/binary-all/Packages) is configured multiple times in /etc/apt/sources.list.d/codelite.list:1 and /etc/apt/sources.list.d/codelite.list:2
W: Target Translations (universe/i18n/Translation-en_US) is configured multiple times in /etc/apt/sources.list.d/codelite.list:1 and /etc/apt/sources.list.d/codelite.list:2
W: Target Translations (universe/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list.d/codelite.list:1 and /etc/apt/sources.list.d/codelite.list:2
W: Target DEP-11 (universe/dep11/Components-amd64.yml) is configured multiple times in /etc/apt/sources.list.d/codelite.list:1 and /etc/apt/sources.list.d/codelite.list:2
W: Target DEP-11-icons (universe/dep11/icons-64x64.tar) is configured multiple times in /etc/apt/sources.list.d/codelite.list:1 and /etc/apt/sources.list.d/codelite.list:2
W: The repository 'http://repos.codelite.org/ubuntu uni Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch http://repos.codelite.org/ubuntu/dists/uni/iverse/binary-amd64/Packages  404  Not Found
E: Some index files failed to download. They have been ignored, or old ones used instead.
W: Target Packages (universe/binary-amd64/Packages) is configured multiple times in /etc/apt/sources.list.d/codelite.list:1 and /etc/apt/sources.list.d/codelite.list:2
W: Target Packages (universe/binary-i386/Packages) is configured multiple times in /etc/apt/sources.list.d/codelite.list:1 and /etc/apt/sources.list.d/codelite.list:2
W: Target Packages (universe/binary-all/Packages) is configured multiple times in /etc/apt/sources.list.d/codelite.list:1 and /etc/apt/sources.list.d/codelite.list:2
W: Target Translations (universe/i18n/Translation-en_US) is configured multiple times in /etc/apt/sources.list.d/codelite.list:1 and /etc/apt/sources.list.d/codelite.list:2
W: Target Translations (universe/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list.d/codelite.list:1 and /etc/apt/sources.list.d/codelite.list:2
W: Target DEP-11 (universe/dep11/Components-amd64.yml) is configured multiple times in /etc/apt/sources.list.d/codelite.list:1 and /etc/apt/sources.list.d/codelite.list:2
W: Target DEP-11-icons (universe/dep11/icons-64x64.tar) is configured multiple times in /etc/apt/sources.list.d/codelite.list:1 and /etc/apt/sources.list.d/codelite.list:2`

Benji15150, 2019-02-08:
On the side note, I have recently recloned airsim, and the older version of airsim without multi-vehicle attribute opened just fine without that warning. I am not sure what is causing missing compiler error for this version of block.uproject.

msb336, 2019-02-19:
Try cleaning out your repository and rebuilding. The first step would be to run ```./clean.sh``` inside the blocks project directory, and then trying to open the environment again. If that doesn't work, you can clear out the entire repo by running ```git clean -fdx```. Be warned, this will remove every file not commited to the repository, so if you have any personal work in the repo, make sure it is either committed to your fork or saved somewhere else on your machine.

msb336, 2019-03-20:
If you are still having a problem related to this issue, please re-open it

closed 2019-03-20 by msb336