microsoft/AirSim #1781: Simulation without rendering

Ahmad5112, 2019-02-09:
# Read This First

I am trying to run the simulation in batch mode. I am using unreal engine 4.18 but after sometimes it stops.I  want to run a simulation without rendering graphics 
as narrated in issue #850 I have set view mode to no display but it is still rendering as it is taking a lot of time and i want to save time and resources ?is it possible if so how it is?

madratman, 2019-02-19:
- Set the following two environment variables:
```shell
	$ export SDL_VIDEODRIVER=offscreen
	$ export SDL_HINT_CUDA_DEVICE=0
```
And then try running your package.

closed 2019-02-19 by madratman

Ahmad5112, 2019-02-19:
where these variable need to set?

madratman, 2019-02-19:
just copy paste the commands in your command line, which you can open using ctrl+alt+t

avcuenes, 2020-10-23:
> If you're on linux, set the following two environment variables:
> 
> ```shell
> 	$ export SDL_VIDEODRIVER=offscreen
> 	$ export SDL_HINT_CUDA_DEVICE=0
> ```
> 
> And then try running your package.

Hi I am using windows how can I do it

huyaoyu, 2021-06-03:
For future friends who get stuck on offscreen rendering, headless server, or running binaries in Docker containers, [this](https://github.com/microsoft/AirSim/issues/3703#issuecomment-854221381) might help.