microsoft/AirSim #1785: Segmentation and depth images are darker than editor

MahirGulzar, 2019-02-12:
Hi,

I have been comparing results of segmentation and depth perspective after recording them using HUD record button or 'R' key in UE4. The result of segmentation and depth map is not similar to what editor is showing i.e its dark and in case of Depth map (Perspective) its way too dark that you can't even distinguish near and far object from it.

**The result from editor is as follows:**

[image]

**and recorded images are here:**

[image]
[image]

I have already played around with gamma and exposure and its solving the problem. I guess I don't even need eye adaptation thats why I already turned it off from the project.

**Settings.json has following settings for cameras:**

  "Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.05,
    "Cameras": [
      {
        "CameraName": "0",
        "ImageType": 0,
        "PixelsAsFloat": false,
        "Compress": true
      },
      {
        "CameraName": "0",
        "ImageType": 3,
        "PixelsAsFloat": false,
        "Compress": true
      },
      {
        "CameraName": "0",
        "ImageType": 5,
        "PixelsAsFloat": false,
        "Compress": true
      }
    ]
  },
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 640,
        "Height": 480,
        "FOV_Degrees": 90,
        "TargetGamma": 2.0
      },
      {
        "ImageType": 3,
        "Width": 640,
        "Height": 480,
        "FOV_Degrees": 90,
        "AutoExposureSpeed": 1000,
        "AutoExposureBias": 0,
        "AutoExposureMaxBrightness": 1.00,
        "AutoExposureMinBrightness": 0.03,
        "MotionBlurAmount": 0,
        "TargetGamma": 2.0,
        "ProjectionMode": "",
        "OrthoWidth": 5.12
      },
      {
        "ImageType": 5,
        "Width": 640,
        "Height": 480,
        "FOV_Degrees": 90
      }
    ]
  },

Any help would be highly appreciated.

chaitanyantr, 2019-02-12:
please see the similar issues for depth.

MahirGulzar, 2019-02-12:
> please see the similar issues for depth.
> #921

I already saw the issue but haven't been able to figure out the problem. For example in case of segmentation the colors dont have the same values as the editor i.e they are darker. I followed the same approach of extracting them uncompressed but the pixels values are darker.

MahirGulzar, 2019-02-13:
Ok for anyone who is still facing an issue I finally got the solution for both images:

**[Solution] Depth Map:**

- Make sure pixelAsFloat is set to True in settings.json while recording DepthPerspective image from Unreal. In this case I set images to be uncompressed i.e compressed: False.
- Once the images are extracted in AirSim folder. The depth images will be in .pfm format in order to convert them in .png format with proper depth information I wrote following script.
**NB!:** Fill the highlighted code accordingly

`import numpy as np
import cv2
import glob

images = [cv2.imread(file) for file in glob.glob('C:\**path-to-folder-with-pfm-images**\*pfm')]

for idx, image in enumerate(images):
    depth = np.array(image, dtype=np.float32)
    depth = depth.reshape(**height**, **width**)
    depth = np.flipud(depth)
    cv2.imwrite('depth'+str(idx+1)+'.png', depth)`

**[Solution] Segmentation:**

(Here the record settings in settings.json had PixelToFloat:False and I am using compressed images)

In this case as I was using my own custom pallete i.e I went into the segmentation material in UE4 and switched the pallete with my own. So if you switched the default pallete in Unreal by your own, Make sure you do the following changes: 

- Change following properties of texture file in UE4 as shown below
[image]

- Click on the texture sampler node in segmentation material and make sure its of type linear color as shown below
[image]

Dont forget to recompile the material file before running.

¬Cheers

closed 2019-02-13 by MahirGulzar

lucafei, 2020-12-01:
hi, @MahirGulzar i have a question for the codes. why you need to make a flip operation on the depth?