microsoft/AirSim #1786: How to get "only" left and right images of a stereo camera in Airsim, I want to train(ML) with those images.

chaitanyantr, 2019-02-12:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------
I am able to get the images with 

"client.simGetImages(airsim.ImageRequest("0", airsim.ImageType.DepthVis))"---(i am believing that this use the stereo camera of 25cm base line)

How can i get the only left and only right images of the stereo camera in Multi-rotor.?

madratman, 2019-02-19:
Q. `Are these cameras all stereo or monocular?`:    
- Please see my comment here https://github.com/Microsoft/AirSim/issues/1787#issuecomment-464250107

Q. `How to get "only" left and right RGB images of a stereo camera in Airsim`   
- In Python, 
```Python
response = client.simGetImages([
   airsim.ImageRequest("front_left", airsim.ImageType.Scene, False, False),
   airsim.ImageRequest("front_right", airsim.ImageType.Scene, False, False)
])

img_left_temp = np.fromstring(response[0].image_data_uint8, dtype=np.uint8)
img_left_rgba = img_left_temp.reshape(response[0].height, response[0].width, 4)

img_right_temp = np.fromstring(response[1].image_data_uint8, dtype=np.uint8)
img_right_rgba = img_right_temp.reshape(response[1].height, response[1].width, 4)
```

In C++, you can do a:
```cpp
msr::airlib::MultirotorRpcLibClient client;
typedef msr::airlib::ImageCaptureBase::ImageRequest ImageRequest;
typedef msr::airlib::ImageCaptureBase::ImageResponse ImageResponse;
typedef msr::airlib::ImageCaptureBase::ImageType ImageType;

std::vector<ImageRequest> request = { 
        ImageRequest("front_left", ImageType::Scene, false, false), 
        ImageRequest("front_right", ImageType::Scene, false, false) 
    };
const std::vector<ImageResponse>& response = client.simGetImages(request);
```

Q. `i am believing that this use the stereo camera of 20 cm base line`   
- If you want to change the baseline, please see my comment here https://github.com/Microsoft/AirSim/issues/1770#issuecomment-462949223

closed 2019-02-19 by madratman

hightao, 2019-04-25:
If you want to get only RGB pic, instead of RGBA, you can code like this:

- Python
```     responses = self.simGetImages([airsim.ImageRequest("3", AirSimImageType.Scene)])
        png_image = np.fromstring(responses[0].image_data_uint8, dtype=np.uint8)
        im = cv2.imdecode(png_image,cv2.IMREAD_COLOR)
        cv2.imshow("img", im)
        cv2.waitKey(0) 
```
other para, please refer to doc.