microsoft/AirSim #1788: How to Add New Api Airsim Python Client For Unity

mustfkeskin, 2019-02-15:
Hello 

I want to access each object color,id, mesh render name, layer name from python.
For develop data collection like **Mask-RCNN** format
I change CameraFiltersScript.cs(**UNITY**) to export csv format.
Accessing pallete info and object colors from python client is much more appropriate  than exporting csv i think so how i can add new api?

There is a any guide for develop new api unity-python

Thanks

msb336, 2019-02-15:
@mustfkeskin Unfortunately there is no guide for adding new apis, but I can give you the relevant code to help you get started.

- [Here](https://github.com/Microsoft/AirSim/blob/master/PythonClient/airsim/client.py) is the python client class. It holds all of the python client-side api message types to be sent to the airsim server. You'll ultimately adding a new function with an rpc binding to the function you have built in airsim

- You will need to build your function in airsim, and then bind it to an rpc call on the serverside. [Here](https://github.com/Microsoft/AirSim/blob/master/AirLib/src/api/RpcLibServerBase.cpp) is where functions are bound to incoming rpc requests.

Hopefully this was helpful for you. Happy coding!

mustfkeskin, 2019-02-18:
Thank u @msb336 
 I will try to develop new api
:)

closed 2019-03-20 by msb336