microsoft/AirSim #1789: server could not find function 'getServerVersion' with argument count 0.

BrianN92, 2019-02-15:
After building the latest released version, I failed to run the "hello_drone.py" file with error "server could not find function 'getServerVersion' with argument count 0.". Did anyone run into this error before? How can I fix it? Thanks in advance!

madratman, 2019-02-19:
that's unexpected. Are you sure you're using the 1.2 binaries?
There are a couple of other issues of ppl running into this : https://github.com/Microsoft/AirSim/issues/1179 and https://github.com/Microsoft/AirSim/issues/1189. you can look at the comments therein

what environment and python version are you using? I'd encourage to build airsim again.

jonyMarino, 2020-08-13:
Hi @BrianN92 did you find what was wrong? I think we have to close this issue if you don't provide more info

closed 2020-08-13 by jonyMarino