microsoft/AirSim #1790: [UNSOLVED] - CHANGING THE BASELINE

nalinraut, 2019-02-15:
I am re-opening a new issue since I have no response on the other issue and eventhough this particular issue may be a solved one, none of the solutions are working for me or I might be going wrong somewhere, however, without any response it is getting a bit difficult for me to carry on with my project. 

Following is what I tried for changing baseline:

( a) I edit `BP_FlyingPawn` using UE4Editor as suggested in  https://github.com/Microsoft/AirSim/issues/1770#issuecomment-462949223. 

(b )Next, I save, build, and compile in UE4Editor itself. I close the UE4Editor and reopen it to find no change in the baseline. 

Left Camera in `BP_FlyingPawn`:
[image]

Right Camera in `BP_FlyingPawn`:
[image]

The images retrieved from Front RIght, Front Left, and Front Center, are evident from snap below.:
[image]

Like both Left and Right cameras are 2006 units apart.

The `settings.json` file I used is:
`
{
  "SettingsVersion": 1.2,
  "SimMode":"Multirotor",
  "SubWindows": [
    {"WindowID": 0, "CameraName": "1", "ImageType": 0, "Visible": false},
    {"WindowID": 1, "CameraName": "2", "ImageType": 0, "Visible": false},
    {"WindowID": 2, "CameraName": "0", "ImageType": 0, "Visible": false}   
  ]
}
`

I would really be glad to have this issue solved.

madratman, 2019-02-15:
can you save and compile in the BP window itself? that works for me. You'd need y=-6 and +6 for left and right cams for ZED. 
Y=20000 is def a crazy value, and if got saved, it means you did it correctly once.

nalinraut, 2019-02-16:
@madratman Thank you for the response. I could save and build it in the UE4Editor window, however, when I play without closing Unreal, Unreal just shuts off/ crashes.  And when I reopen it, even though the value is saved, it is not being used. Also, at y = 20000, I get to see the same image as seen at y = 25. The values are definitely being saved but not used.

madratman, 2019-03-11:
My earlier answer was way too complicated. 
Baseline can be changed from settings.json itself. 
Instead of centimeters, use meters. So it will be Y=-0.06, and Y=+-0.06

```json
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "ClockSpeed": 1.0,
  "SubWindows": [
  {"WindowID": 0, "ImageType": 0, "CameraName": "front-left", "Visible": true},
  {"WindowID": 2, "ImageType": 0, "CameraName": "front-right", "Visible": true}
  ],
  "Vehicles": {
    "SimpleFlight": {
      "VehicleType": "SimpleFlight",
      "DefaultVehicleState": "Armed",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "RC": {
        "RemoteControlID": 0,
        "AllowAPIWhenDisconnected": false
      },
      "Cameras": {
        "front-left": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90
            }
          ],
          "X": 0.50, "Y": -0.06, "Z": 0.10,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        },
        "front-right": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 672,
              "Height": 376,
              "FOV_Degrees": 90
            }
          ],
          "X": 0.50, "Y": 0.06, "Z": 0.10,
          "Pitch": 0.0, "Roll": 0.0, "Yaw": 0.0
        }
      },
      "X": 0, "Y": 0, "Z": 0,
      "Pitch": 0, "Roll": 0, "Yaw": 0
    }
  }
}```

nalinraut, 2019-03-15:
@madratman Thanks! This works.

closed 2019-03-15 by nalinraut