microsoft/AirSim #1791: 3D Bounding box

Div99, 2019-02-15:
Hi, can the PythonClient be extended to have a getBoundingBox3D method for other cars, pedestrians in the environment? 

This functionality will go a long way in making AirSim a more complete simulator, allowing for functionality such as perception of objects and prediction of their trajectories, which are standard components of most AV system pipelines.

madratman, 2019-02-19:
Yes, technically it can be, but we don't have the bandwidth to add this as of now. 
One thing to look at is how carla does it here https://github.com/carla-simulator/carla/search?q=bounding+box&type=Code and see if you can port it to airsim.

Div99, 2019-02-19:
Hi, I want to use the feature with pre-built environments like CityEnviron or AirSimNH. From what I understand, any changes I make won't work with the supplied binaries of environments.

ducmanhnguyen, 2020-01-13:
Have anyone success to get 3d bounding box with lidar data for deep learning? It would be great if you can share the data or the code to generate it. Thanks :)

ironclownfish, 2020-02-04:
In the meantime here's a ridculous hack:
You can measure the radius of an object by running it into a wall and noting the object's position relative to the position of the wall. Do this in all 6 directions and you have a bounding box. (This process could be automated, client side)

It would be much better for us to add the feature of course.
I just always try to suggest it's possible to do things client side, even if in a very weird way.

saihv, 2020-02-05:
@ironclownfish Is there a way to extract the dimensions of the 'simple' collision box, perhaps?

chongjeelee, 2022-02-11:
hi,guys,does solve this problem for geting 3d bounding box?
if not in airsim api,how can i do something with UE4 bluerprint ?

chongjeelee, 2022-02-16:
i found that we can set lidar and cam on the same position(settings.json),and the cam can get 3d bounding box from environment!so this 3d bounding box can be used to lidar !
maybe can close this issue...