microsoft/AirSim #1793: How to save other sensors information like GPS and IMU?

pengjili, 2019-02-18:
Window, Airsim version:1.83, Unreal version: 4.
Hi, please forgive my stupid question. Now I can successfully run my Airsim. But to do the autonomous driving research I need the IMU and GPS information from the vehicle. For my understanding, it should be easily solved if the settings.json is correctly modified. However I met some problems. My settings.json is like the following:

{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "Recording": {
    "RecordInterval": 0.05,
    "Cameras": [
        { "CameraName": "0", "ImageType": 3, "PixelsAsFloat": false, "Compress": true },
        { "CameraName": "0", "ImageType": 5, "PixelsAsFloat": false, "Compress": true },
        { "CameraName": "0", "ImageType": 0, "PixelsAsFloat": false, "Compress": true }
    ]
  },
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 600,
        "Height": 300

      },
      {
        "ImageType": 5,
        "Width": 600,
        "Height": 300

      },
      {
        "ImageType": 3,
        "Width": 600,
        "Height": 300

      }
    ]
  },
  "SubWindows": [
    {"WindowID": 0, "CameraName": "0", "ImageType": 3, "Visible": false},
    {"WindowID": 1, "CameraName": "0", "ImageType": 5, "Visible": false},
    {"WindowID": 2, "CameraName": "0", "ImageType": 0, "Visible": false}    
  ],
  "DefaultSensors": {
        "Barometer": {
             "SensorType": 1,
             "Enabled" : true
        },
        "Gps": {
             "SensorType": 3,
             "Enabled" : true
        },
	"Imu": {
             "SensorType": 2,
             "Enabled" : true
        },
        "Lidar1": { 
             "SensorType": 6,
             "Enabled" : true,
             "NumberOfChannels": 16,
             "PointsPerSecond": 10000
        }
    }
}

I did like this but cannot save any GPS and IMU information. Please tell me what should do to get them, thanks so much!!

pengjili, 2019-02-18:
And I hope the IMU and GPS information can be save in airsim_rec.txt file. Is that possible?

closed 2019-03-15 by pengjili

meitiever, 2019-12-09:
have you solved this problem?