microsoft/AirSim #1796: How do I connect MavLinkTest to the Simulator

kaioforte, 2019-02-19:
I've followed all the tutorial from https://github.com/Microsoft/AirSim/blob/master/docs/px4_sitl.md to run PX4 in SITL mode on Windows 10 and everything works fine. Then I tried to run MavLinkTest app with my simulaton so I could control the drone with the commands from the app, but when I start the MavLinkTest (selecting it as Startup Project and hitting F5 on VS2017) I receive this error: **" Must specify one of -serial, -udp or -server otherwise we don't have a drone connection "** and the prompt automatically closes leaving no chance to insert any command . So I wanna know where I could setup this connections?
Like in https://microsoft.github.io/AirSim/docs/px4_logging/
I looked at the documentation but it's only tells me the command I should use but not where.
I'm using a Windows 10 machine, Unreal Engine 4.18, VS2017 and the latest version of AirSim.

I aprecciate if someone could help me, I'm new on this "AirSim World".

kaioforte, 2019-03-18:
Up

shantanusingh16, 2019-09-21:
I have a similar issue. I'm trying to use the MavLinkTest to intercept the messages from pixhawk to airsim running in HITL mode. However, I'm unable to get it to work properly as AirSim keeps showing the message that 'No messages received from HIL node. Restart HIL node'. 

I'm following the instructions on this page and I'm able to connect to QGroundControl via UDP to Pixhawk via MavLinkTest. However, it doesn't work for AirSim.

Any ideas?

lovettchris, 2021-06-04:
Yeah, sorry currently it is not possible.  mavlinktest would need to be modified so it can connect to -serial port and setup a UDP server that AirSim can connect to.  This is close:
```
MavLinkTest.exe -serial -proxy:localhost:14580
```  
But doesn't quite work because this is a broadcast, not a connection oriented UDP channel.  This is what we'd want:
```
MavLinkTest.exe -serial -server -udp:localhost:14580
```
but then you'll get an error that this configuration isn't supported...   but a pull request would be welcome :-)