microsoft/AirSim #1798: Launching script from within AirSim

seanny1986, 2019-02-20:
I'd like to use AirSim for RL training, and I've managed to set up a basic gym environment to do what I need (Python 3.5). Ideally, I'd like to be able to run the training algorithm faster than real time; so far, I've figured out how to speed AirSim up using the settings.json file, but the problem is that if I run AirSim and speed it up, the training script is running separately from a terminal at its own speed. If I'm running AirSim at 5x real time, the RL algorithm is still running at the same speed, and is now effectively sending commands to the aircraft 5x slower. 

Is there any way I can launch the training algorithm from within AirSim so that this doesn't happen?

seanny1986, 2019-02-25:
Bump. Any help here would be greatly appreciated.

msb336, 2019-03-29:
There is no way to have the training algorithm running inside of airsim without significant modification to the source code.

However, you can scale your RL script update frequency to the accelerated AirSim simulation speed. Obviously, the issue here is that your real time update frequency will be 5x slower than your max possible update frequency on your machine.

For example, if your RL algorithm operates at a max frequency of 20 hz and you have the training setup running at 5x normal speed, you can drop the update frequency to 4 hz when returning to 1x real time in sim.

closed 2019-10-08 by msb336