microsoft/AirSim #1800: Accesing Unreal Engine's Mesh Distance Fields

madratman, 2019-02-20:
Does anyone have experience with https://docs.unrealengine.com/en-us/Engine/Rendering/LightingAndShadows/MeshDistanceFields? I am unable to find an API for the same (unless I am not googling this properly), however as evident from the above section in unreal's documentation and corresponding videos for lighting and shadows for rendering in UE4, it builds the distance field anyway. Question is, how can we query it?

It would be really useful to be able to query the distance field programatically, for autonomy related tasks like trajectory optimization / collision avoidance. Also, it serves as a ground truth for mapping algorithms

madratman, 2019-02-20:
ok, seems like meshdistancefield is the wrong keyword to look at. 
there's [FDistanceFieldVolumeData](http://api.unrealengine.com/INT/API/Runtime/Engine/FDistanceFieldVolumeData/index.html) [FDistanceFieldVolumeTexture](http://api.unrealengine.com/INT/API/Runtime/Engine/FDistanceFieldVolumeTexture/index.html) for a static mesh

ironclownfish, 2020-02-04:
Ask Sai

GuangyaoSHI, 2021-02-24:
> ok, seems like meshdistancefield is the wrong keyword to look at.
> there's [FDistanceFieldVolumeData](http://api.unrealengine.com/INT/API/Runtime/Engine/FDistanceFieldVolumeData/index.html) and [FDistanceFieldVolumeTexture](http://api.unrealengine.com/INT/API/Runtime/Engine/FDistanceFieldVolumeTexture/index.html) for a static mesh
> 
> the above two are wrt to a single static mesh. There's also a `Global Distance Field` (ctrl+f "Global Distance Field" in the [MeshDistanceFields doc](https://docs.unrealengine.com/en-us/Engine/Rendering/LightingAndShadows/MeshDistanceFields). That's what we really need

Do you know how to query distance field now from Unreal? I am doing a project and I want to use that information for trajectory optimization.

rajat2004, 2021-02-24:
Seems to be somewhat similar to the newly added [Voxel Grid API](https://microsoft.github.io/AirSim/api_docs/html/#airsim.client.VehicleClient.simCreateVoxelGrid), see the original PR https://github.com/microsoft/AirSim/pull/3209 and the docs update PR https://github.com/microsoft/AirSim/pull/3239