microsoft/AirSim #1803: Is there unreal project file other than Blocks for linux?

Benji15150, 2019-02-21:
Hello, 

I was wondering whether there is other u.project file other than blocks environment that can be run in linux. I downloaded the airsimNH, but it seems like I can't create unreal project with it. Is there a way to run a simulation without creating a project? or is there other unreal project file that is not a block environment?

Thanks!

msb336, 2019-02-22:
@Benji15150 
Unfortunately not, but that is due to licensing, and not isolated to the linux platform. We can't share the editor files to our other environments because they contain 3rd party products that are not able to distribute freely.

However, the airsim plugin is cooked into the release binaries listed [here](https://github.com/Microsoft/AirSim/releases/tag/v1.2.0Linux), so you are still able to run simulations. The only thing you cannot do is modify the sim environment.

closed 2019-02-22 by msb336

eds89, 2019-05-09:
Hi. So does that mean I cannot create a new post process material in the environment?