microsoft/AirSim #1805: Can I hide the mesh of the drone in Unreal?

whatseven, 2019-02-22:
In the drone mode of airsim. When I request an `image` in some specific `angle`, I will get an image `include the mesh of the drone(see below)`. But in real time, my camera will never capture the drone itself in any angles. So I' m wondering if there is a solution or setting to `prevent the camera to render the drone mesh` when it is capturing an image.

I found the setting in both unreal editor details panel and drone blueprint but no result. I search on google and found a similar question. It says that you can turn on the `owner no see`. But in airsim it seems that the drone mesh is the camera's owner.

[image]

saihv, 2019-10-30:
Perhaps an easier fix for this would be to move the camera a bit to the front, so that it doesn't capture the mesh.

jonyMarino, 2020-08-14:
I understand that for your question Sai's answer is sufficient and does not have to be done otherwise. Feel free to reopen this issue if you don't think so.

closed 2020-08-14 by jonyMarino

icisneros, 2021-01-23:
Is it possible to change the ownership of the external camera that follows the drone?  I am looking to make the drone invisible to the drone's cameras, but visible to the player (or pilot).  Allowing the external camera to not be owned by the FlyingPawn would allow me to set the "Only Owner See" setting on the drone mesh.  My current setup is a set of six cameras all situated at a point a few inches below the drone.  The solution proposed by saihv is possible (would require relocating the six drone cameras such that the drone is not in their FOV), but ultimately unsatisfactory for me because it results in the cameras producing slightly offset images from each other.