microsoft/AirSim #1806: Coordinates transformation from opencv camera frame to AirSim NED for object localization

marcociara379, 2019-02-22:
Hi everyone,
I am trying to detect an Aruco marker in AirSim and localize it in NED frame after detecting it in the camera frame.
To check the correctness of my computations, I am comparing the estimated pose in quaternions with respect to the ground truth output by AirSim. Unfortunately, it seems that I have did something wrong and the two values mismatch. 
Here is my procedure:

1. I compute the `rvec`, `tvec` rotation and translation vectors with  `cv2.aruco.estimatePoseSingleMarkers(corners, aruco_length, camera_K_mat, dist_coeffs)`. These should refer to the *transformation from the Aruco frame to the OpenCV camera frame*, that is *the pose of the aruco marker in camera coordinates* (cam frame should be x=east; y=down; z=north; aruco frame is x:east, y:north, z=out of the marker).
2.  I convert the `rvec` rotation vector in rotation matrix: `R_aruco_cam_, _ = cv2.Rodrigues(rvec)`
3. I swap the axis of the `R_aruco_cam_` matrix in order to get the aruco position in NED (from camera EDN): `R_aurco_ned = np.array([R_aruco_cam_[:,2], R_aruco_cam_[:,0], R_aruco_cam_[:,1]])`
4. I convert the rotation matrix `R_aurco_ned` to quaternions 

The aruco marker pose from `client.simGetObjectPose()` does return a different orientation, unfortunately.
The quaternion conversion should be correct.

ShrutheeshIR, 2020-08-03:
Hi @marcociara379 
Were you able to solve this issue? I am facing a kind of similar issue. Does simply swapping the axes do, or I'm not sure if a negative sign comes somewhere. Let me know what you did to solve this issue
Thanks!

nakole2, 2020-11-02:
> Hi @marcociara379
> Were you able to solve this issue? I am facing a kind of similar issue. Does simply swapping the axes do, or I'm not sure if a negative sign comes somewhere. Let me know what you did to solve this issue
> Thanks!

Hi @ShrutheeshIR,  did you find the solution for it? I am stuck on the same problem.

closed 2021-11-30 by jonyMarino