microsoft/AirSim #1809: Steering Wheel Angle, Throttle and Brake values incorrect when connected with Logitech G920.

tfromb, 2019-02-26:
I've connected my Logitech G920 Steering wheel and followed the 'Connecting a Steering Wheel' instructions. When the steering wheel is connected, the steering angle starts at 0.5, thus turns right constantly and the acceleration and brake values start at 1.0 - results in the car being unable to move. When I press the acceleration pedal, the value decreases from 1.0 to 0.3 (as if braking). 

I've looked at the CarPawn.cpp however when I change values as instructed in the documentation, nothing is affected. Where is the code that corresponds to these values? 

I've built on a Windows 10 laptop. 

Thank you

wintersoom, 2019-07-09:
i have the same problem.do you solve it?

jonyMarino, 2020-08-14:
@tfromb any update on this issue?

lemonade1104, 2021-04-15:
actually you can control the offset and scale directly within Unreal Project Setting page. It will solve you problem i suppose.
But now I have another issue: the brake pedal value is not perfectly zero, always like "0.003" or something, which will make the car do not move. I can only drive the car when i set the scale as "0" to make sure the brake pedal value is pefect value, so that i can drive.  so sad.