microsoft/AirSim #1811: Unreal engine breaks down when I set a camera model in multirotor mode

didwjdrl12, 2019-02-27:
Unreal engine is shut down when I set a camera model in multirotor mode. 

Unreal engine version: 4.18.3
Airsim commit: 9e919a8
My setting.json
{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
    "FishEyeDrone": {
      "VehicleType": "SimpleFlight",
      "Cameras": {
        "front-center": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "FOV_Degrees": 120
            }
          ]
        }
      }
    }
  }
}
(I mimicked a setting example in the settings.md)

When I click the "Play button" in the engine, the engine has crashed. 
[image]
This is a screenshot.

Log file open, 02/27/19 16:44:15
LogWindows: Failed to load 'aqProf.dll' (GetLastError=126)
LogWindows: File 'aqProf.dll' does not exist
LogWindows: Failed to load 'VSPerf140.dll' (GetLastError=126)
LogWindows: File 'VSPerf140.dll' does not exist
LogWindows: Failed to load 'VtuneApi.dll' (GetLastError=126)
LogWindows: File 'VtuneApi.dll' does not exist
LogWindows: Failed to load 'VtuneApi32e.dll' (GetLastError=126)
LogWindows: File 'VtuneApi32e.dll' does not exist
LogInit: Display: Running engine for game: Blocks
LogPlatformFile: Not using cached read wrapper
LogInit: Display: RandInit(-1188731880) SRandInit(-1188731880).
LogInit: Read version file with build ID '3709383'
LogInit: Overriding engine version to CL 3832480, compatible CL 3709383
LogInit: Registered custom module enumerator with build ID '3709383'
LogTaskGraph: Started task graph with 5 named threads and 32 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.382664
LogD3D11RHI: Aftermath initialized
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin CodeLiteSourceCodeAccess
LogPluginManager: Mounting plugin GitSourceControl
LogPluginManager: Mounting plugin KDevelopSourceCodeAccess
LogPluginManager: Mounting plugin NullSourceCodeAccess
LogPluginManager: Mounting plugin PerforceSourceControl
LogPluginManager: Mounting plugin SubversionSourceControl
LogPluginManager: Mounting plugin UObjectPlugin
LogPluginManager: Mounting plugin VisualStudioCodeSourceCodeAccess
LogPluginManager: Mounting plugin VisualStudioSourceCodeAccess
LogPluginManager: Mounting plugin XCodeSourceCodeAccess
LogPluginManager: Mounting plugin AssetManagerEditor
LogPluginManager: Mounting plugin AudioCapture
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin GameplayTagsEditor
LogPluginManager: Mounting plugin MobileLauncherProfileWizard
LogPluginManager: Mounting plugin PluginBrowser
LogPluginManager: Mounting plugin SpeedTreeImporter
LogPluginManager: Mounting plugin AlembicImporter
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin HTML5Networking
[... 375 lines cut ...]
[2019.02.27-07.50.33:256][728]LogOutputDevice: Error: [Callstack] 0x000000006E42DAC1 UE4Editor-Engine.dll!UnknownFunction []
[2019.02.27-07.50.33:256][728]LogOutputDevice: Error: [Callstack] 0x000000006EEC06AE UE4Editor-Engine.dll!UnknownFunction []
[2019.02.27-07.50.33:256][728]LogOutputDevice: Error: [Callstack] 0x000000006E4504CE UE4Editor-Engine.dll!UnknownFunction []
[2019.02.27-07.50.33:256][728]LogOutputDevice: Error: [Callstack] 0x000000006C84B235 UE4Editor-UnrealEd.dll!UnknownFunction []
[2019.02.27-07.50.33:256][728]LogOutputDevice: Error: [Callstack] 0x000000006C87556E UE4Editor-UnrealEd.dll!UnknownFunction []
[2019.02.27-07.50.33:256][728]LogOutputDevice: Error: [Callstack] 0x000000006C89192F UE4Editor-UnrealEd.dll!UnknownFunction []
[2019.02.27-07.50.33:256][728]LogOutputDevice: Error: [Callstack] 0x000000006C2E6024 UE4Editor-UnrealEd.dll!UnknownFunction []
[2019.02.27-07.50.33:257][728]LogOutputDevice: Error: [Callstack] 0x000000006CB94AC6 UE4Editor-UnrealEd.dll!UnknownFunction []
[2019.02.27-07.50.33:257][728]LogOutputDevice: Error: [Callstack] 0x00000000582E5A26 UE4Editor.exe!UnknownFunction []
[2019.02.27-07.50.33:257][728]LogOutputDevice: Error: [Callstack] 0x00000000582F5430 UE4Editor.exe!UnknownFunction []
[2019.02.27-07.50.33:257][728]LogOutputDevice: Error: [Callstack] 0x00000000582F54AA UE4Editor.exe!UnknownFunction []
[2019.02.27-07.50.33:257][728]LogOutputDevice: Error: [Callstack] 0x0000000058302379 UE4Editor.exe!UnknownFunction []
[2019.02.27-07.50.33:257][728]LogOutputDevice: Error: [Callstack] 0x0000000058303D57 UE4Editor.exe!UnknownFunction []
[2019.02.27-07.50.33:257][728]LogOutputDevice: Error: [Callstack] 0x00000000D27781F4 KERNEL32.DLL!UnknownFunction []
[2019.02.27-07.50.33:257][728]LogOutputDevice: Error: [Callstack] 0x00000000D559A251 ntdll.dll!UnknownFunction []
[2019.02.27-07.50.33:257][728]LogOutputDevice: Error: 
[2019.02.27-07.50.33:267][728]LogStats:                SubmitErrorReport -  0.000 s
[2019.02.27-07.50.33:720][728]LogStats:                    SendNewReport -  0.454 s
[2019.02.27-07.50.33:720][728]LogStats:             FDebug::EnsureFailed -  1.331 s
[2019.02.27-07.50.33:721][728]LogSpawn: Warning: SpawnActor failed because the given transform (nan,nan,-nan(ind)|-90.000000,0.000000,-0.000000|1.000000,1.000000,1.000000) is invalid

This is log.

I can't set stereo cam because of this error. I will appreciate if you fix this error as soon as possible.
Thanks.

jonyMarino, 2020-08-20:
@didwjdrl12 do you still have this issue on latest airsim version? The current recommended UE version is 4.24

analytic-bias, 2025-03-15:
Same issue encountered with [Colosseum](https://github.com/CodexLabsLLC/Colosseum); documenting it here for others in the future encountering this. The issue is due to `PawnSimApi::createCamerasFromSettings() [.../Unreal/Environments/BlocksV2 5.4/Plugins/AirSim/Source/PawnSimApi.cpp]` at the following lines
```cpp
FTransform camera_transform(FRotator(setting.rotation.pitch, setting.rotation.yaw, setting.rotation.roll),
                                    position,
                                    FVector(1., 1., 1.));

...

camera->AttachToComponent(bodyMesh, FAttachmentTransformRules::KeepRelativeTransform);
```
where the default values for `setting.rotation.pitch` etc. are all `NaN` if not specified in `settings.json`. The printed call stack suggested the second line above was where the seg fault happened, then further in `AActor::AttachToComponent(USceneComponent*, FAttachmentTransformRules const&, FName)`. It also prompted about said `NaN` quaternion
```
[2025.03.15-07.21.50:014][842]LogCore: Error: Invalid input P=nan Y=nan R=nan to FRotator::Quaternion - generated NaN output: X=nan Y=-nan Z=-nan W=nan
```

----

A temporary fix should be trivial: in `settings.json`, use
```json
{
    "SettingsVersion": 1.2,
    "SimMode": "Multirotor",
    "Vehicles": {
        "PX4": {
            "VehicleType": "PX4Multirotor",
            "Cameras": {
                "0": {
                    "CaptureSettings": [
                        {
                            "ImageType": 0,
                            "Width": 255,
                            "Height": 255,
                            "FOV_Degrees": 120
                        }
                    ],
                    "X": 0, "Y": 0, "Z": 0,
                    "Pitch": 0, "Roll": 0, "Yaw": 0
                }
            },
            ...
        }
    }
}
```
while a more permanent solution would be to, e.g. in `PawnSimApi::createCamerasFromSettings`, check for NaN before passing as Euler angles.