microsoft/AirSim #1813: How to duplicate multiple drones?

MarkZhou0113, 2019-02-27:
I'm trying to implement multiple drones in the environment, but when I drag BP_FlyingPawn in the scene and click play, the error message window pop out ->"Error at startup: VehicleSetting for vehicle name BP_FlyingPawn_2 was requested but not found" and unreal engine shutdown after click the OK button. 

I have also tried [https://microsoft.github.io/AirSim/docs/multi_vehicle/]() to modify settings.json but nothing happened. Is there any solution to solve this problem?

*AirSim is the latest version, Unreal version is 4.18.3.

saihv, 2019-02-28:
Does your settings.md file match what you want? For example, for two drones, my settings file looks like this: (note the "Vehicles" part)

```
{
	"SettingsVersion": 1.2,
	"SimMode": "Multirotor",
	"CameraDefaults": {
      "CaptureSettings": [
        {
          "ImageType": 0,
          "Width": 640,
          "Height": 480,
          "FOV_Degrees": 90,
          "AutoExposureSpeed": 30,
          "MotionBlurAmount": 0,
					"TargetGamma": 0.7
        }
    ]
  },
	"Vehicles": {
		"Drone1": {
		  "VehicleType": "SimpleFlight",
		  "X": 0, "Y": 0, "Z": -4,
      "Yaw": 90
    },
    "Drone2": {
		  "VehicleType": "SimpleFlight",
		  "X": 3, "Y": 0, "Z": -4,
      "Yaw": 90
    }
  }
}
```

With this file, for instance, you won't need to drag and drop multiple drone instances.

MarkZhou0113, 2019-03-01:
@saihv Thank you for the reply, I figure out what I did wrong before. I haven't noticed that settings.json file is directly connect to the environment and I just modify the wrong file with same name, which is not in the path. But is it normal that cannot drag drone instances into the scene?

msb336, 2019-05-14:
@MarkZhou0113 
Drone instances are loaded dynamically into the scene via the settings file. If you want to use vehicles that are already in the scene, you will need to set up [pawn paths](https://github.com/Microsoft/AirSim/blob/master/docs/settings.md#pawnpaths) as appropriate.

closed 2019-05-14 by msb336