microsoft/AirSim #1818: Improve image capture speed

tiax615, 2019-03-01:
Hi community，I think I may have found a way to improve the speed of image capture.
Modify the code in PIPCamera.cpp, updateCaptureComponentSetting()，
change
`render_target->InitAutoFormat(setting.width, setting.height);`
to
`render_target->InitCustomFormat(setting.width, setting.height PF_B8G8R8A8, true);`

AirSim use SceneCaptureComponent2D SceneCapture mode is LDR, using RGBA8 is enough, may not need to use RGBA16f, which is supporting HDR mode. The Format of RenderTarget initialized by InitAutoFormat() is RGBA16f, which takes up resources. You can customize the Format by using InitCustomFormat(). So screenshots might become more efficient.

Thank you for reading！

tiax615, 2019-03-01:
By the way, I used the Victory plugin by Rama to take screenshots and the frame rate was up to 20fps!
I think it's also possible to write methods in C++ code without blueprints
[image]

pengjili, 2019-03-08:
@tiax615 Really appreciate your sharing. I've tried your first way(didn't try 2nd yet), it did save a lot of time. But can I ask what's your image's resolution? Mine is 1280x800, and even using viewMode: "NoDisplay",  but it still costs about 250ms. I'd also like to put my frame rate up to 20fps, is there any other way to reduce time on saving images? Thanks so much again

tiax615, 2019-03-13:
> @tiax615 Really appreciate your sharing. I've tried your first way(didn't try 2nd yet), it did save a lot of time. But can I ask what's your image's resolution? Mine is 1280x800, and even using viewMode: "NoDisplay", but it still costs about 250ms each frame. I'd also like to put my frame rate up to 20fps, is there any other way to reduce time on saving images? Thanks so much again

Hi, My image resolution is 1280x720. And I'm looking for a faster way to do it, like video live. OBS?

closed 2019-10-08 by msb336