microsoft/AirSim #1820: Linux binaries crashed when using ComputerVision SimMod

songjin321, 2019-03-03:
## I try to use pre-release Linux binaries Neighborhood in my Ubuntu 16.04 pc, but I failed to run it.
### I edit settings.json file to use ComputerVision SimMode 
```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "ComputerVision"
}
```
### Then I want use Neighborhood enviorment , I run AirSimNH.sh in a terminal, but unfortunately, I get an error as shown below.
[image]

### I want to kown how can I use Linux binaries Neighborhood with ComputerVision SimMode

msb336, 2019-03-20:
@RVBldr This may be an error with the executable file. Unreal can be a bit tricky about including all of the necessary assets from airsim. If that's the case we'll recompile the binary and release the new one as soon as we can.

howard-mahe, 2019-03-21:
I'm facing the same error as well.

msb336, 2019-04-11:
@songjin321 

Just released the updated binary. Please close if your issue has been solved.

songjin321, 2019-04-15:
The updated binary solved the problem, Thank you!

closed 2019-04-15 by songjin321