microsoft/AirSim #1823: Drone controller support in unity

LevNetanel, 2019-03-05:
Hello,

I want to control the Multirotor using the FrSky 2.4GHz ACCST TARANIS X9D PLUS controller.
I followed the guides and succeeded in the blocks environment in Unreal.

But I am not able to control the Multirotor in Unity (However I am able to control the car using the keyboard). I didn't find any tutorial about this. Is Multirotor control supported in Unity?

Running on windows 10 64 bit, Latest Airsim (How can I find the version of Airsim?), Unity 2018.2.15f1.

I tried to modify settings.json and use different "RemoteControlID" values.
This is the current version:

> {
 "SimMode" : "", 
 "SettingsVersion" : 1.2, 
 "SimMode": "Multirotor",
 "Vehicles": {
 	"SimpleFlight": {
 		"VehicleType": "SimpleFlight",
 		"RC": {
 		  "RemoteControlID": 1
 		}
 	}
 },
 "SeeDocsAt" : "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md"
}

Thanks.

vxgu86, 2019-05-13:
problem solved or not?
same problem met.

LevNetanel, 2019-05-13:
I ended up implementing a keyboard controller using python API, and this was OK for my needs.