microsoft/AirSim #1827: Windridge City "ToUnixTimeMilliseconds" Error

chriskim1015, 2019-03-06:
# ToUnixTimeMilliseconds Error

Hi, I have been trying to implement the Drone Demo to the Windridge city environment using this [link](https://github.com/Microsoft/AirSim/blob/master/docs/custom_unity_environments.md). I can successfully run the drone demo in airsim itself using a python API, but I could not implement this in the Windridge City environment.  I have been keep getting an error which says

- Assets/AirSimAssets/Scripts/Utilities/DataManager.cs(63,42):  error CS1061: Type `System.DateTimeOffset' does not contain a definition for `ToUnixTimeMilliseconds' and no extension method `ToUnixTimeMilliseconds' of type `System.DateTimeOffset' could be found. Are you missing an assembly reference?

I found out that this may be a VS 2017 NET compatibility issue, but if I update my visual studio to the newest version, I cannot run Airsim.sln successfully. Does anyone know how to fix this error?

fwarmuth, 2019-03-26:
Same here, did you find the problem?

howard-mahe, 2019-03-27:
`ToUnixTimeSeconds` [was introduced in .NET 4.6](https://docs.microsoft.com/en-us/dotnet/api/system.datetimeoffset.tounixtimeseconds?redirectedfrom=MSDN&view=netframework-4.7.2#applies-to) so the projects needs to match this requirement. I did not find out in which config file one should add this requirement. So I come up with modifying a bit the file `DataManager.cs` from
```cs
public static long GetCurrentTimeInMilli() {
    return DateTimeOffset.UtcNow.ToUnixTimeMilliseconds();
}
```
to:
```cs
private static readonly DateTime UnixEpoch = new DateTime(1970, 1, 1, 0, 0, 0, DateTimeKind.Utc);

public static long GetCurrentTimeInMilli() {
    return (long)(DateTime.Now.ToUniversalTime() - UnixEpoch).TotalMilliseconds;
}
```

wajeehulhassanvii, 2019-04-23:
wow worked like a charm @howard-mahe :)