microsoft/AirSim #1830: Sometime can't take off && always collision with other mesh

whatseven, 2019-03-07:
This makes me confused a long time. Sometimes my drone can't take off successfully. It seems that it always collision with my floor. I' m not sure if it is related to my collision settings.

Here is an gif about it

[image]

whatseven, 2019-03-08:
I found it may be related to the collision info. I have two other gif about it

First, the drone didn't hit anything but the collision info shown on the left screen is keep increasing.

[image]

Second, the drone also didn't hit anything but the collision info shown on the left screen is keep increasing, and the position of the drone was forced to adjust to the cylinder.

[image]

machenxiang, 2019-03-13:
we also meet the same question. In HITL  Simulation,sometimes we can't takeoff the drone. Even the drone takeoff successfully ,the drone seem crazy.It can't hover，constant rising and shaking.

lovettchris, 2019-04-07:
I found the problem with PX4 1.8.2 that stops takeoff from working.  They removed a module that is needed for HITL mode to work, namely, "pwm_sim_out" and so I built a new version of the v1.8.2 with this module included and takeoff and normally flying works again. See [OneDrive download](https://1drv.ms/f/s!AoS-ymEfLuCLqudxN4NCgYtf0f3WPQ).

closed 2020-08-17 by jonyMarino