microsoft/AirSim #1831: Building Airsim on linux offline machine

oliverengineer, 2019-03-09:
Hi everyone.
I am trying to build Airsim on an offline Ubuntu 18 machine.
What i did before is build UE4.21 on Ubuntu 18 online machine and got it working on the offline machine.
Then i built Airsim on the online machine and transferred the compiled Airsim directory to the offline machine , and i considered installing the dependencies on the offline machine too.
But after doing all of this, Airsim didn't compile right on the offline machine and i keep getting errors when starting the setup.sh script.
Is there any guide for a step by step of installing airsim in offline mode?
And when the binary file for linux will be available?
As for the errors, like i said there seem to be errors in dependencies, so what i want to know is what should i consider when building the code in this case?
And thank you.

msb336, 2019-03-21:
@oliverengineer can you copy and paste your error messages into this thread?

oliverengineer, 2019-03-22:
Hi @msb336 and thank you for replying.
Actually i am not getting an error when build the code, rather i am getting an error when running any project containing the airsim plugin on unreal engine.I get the error saying : " The following modules are missing...".
I am using ue4.21, but i found that it is recommended to use 4.18, so i am trying now to install 4.18 version to see if it will work or not.
I will be post later the result of this test to see if it will change anything.

msb336, 2019-04-02:
@oliverengineer did rolling back to `4.18` solve your issue?

msb336, 2019-05-14:
closing due to inactivity. please re-open if you are still having trouble.

closed 2019-05-14 by msb336