microsoft/AirSim #1837: Too slow for recorded images on UE4.18

pengjili, 2019-03-13:
Now I'm trying every way that I know to increase the fps for recorded images(1280x800), Now the fastest I can get is about 10hz. First I will list out all my changes that I made:
 1. Set "ViewMode" : "NoDisplay"
 2. Only record  scene images which means that I turned off my depth view and Segmentation view sub-windows.
 3. Change compressed image format to JPEG(default is png)
 4. `render_target->InitAutoFormat(settings.width, settings.height);` changed to 
     `render_target->InitCustomFormat(settings.width, settings.height, PF_B8G8R8A8, true);`
 
Now I know most of the time cost is coming from `RHICmdList.ReadSurfaceData(),` but just have no idea how to solve it. And from the materials that I've collect I know in the previous version of Airsim and UE doesn't go this slow, right? Since it's not proper to downgrade any one of them, Is there any other way to achieve 50+FPS? 
My setup:
Windows10, Nvidia Quadro K1100M, UE4.18, Airsim latest version

hesham-akmal, 2019-03-29:
Give unity version a try

pengjili, 2019-04-09:
@hesham-akmal Yes I did try unity version and it's indeed very fast. But unreal version has so many amazing features that I'm working on, so I still need my unreal go faster. Anyway thanks for your advice.