microsoft/AirSim #1840: DQNCar.py returns error on image reshaping

hrshovon, 2019-03-15:
Airsim version: 1.2

Python file: DQNcar.py

Whenever I try to run this code, I get this error:

```
Traceback (most recent call last):
  File "DQNcar.py", line 513, in <module>
    current_state = transform_input(responses)
  File "DQNcar.py", line 434, in transform_input
    img2d = np.reshape(img1d, (responses[0].height, responses[0].width))
  File "C:\Users\shovon\AppData\Local\conda\conda\envs\ml_agents\lib\site-packages\numpy\core\fromnumeric.py", line 257, in reshape
    return _wrapfunc(a, 'reshape', newshape, order=order)
  File "C:\Users\shovon\AppData\Local\conda\conda\envs\ml_agents\lib\site-packages\numpy\core\fromnumeric.py", line 52, in _wrapfunc
    return getattr(obj, method)(*args, **kwds)
ValueError: cannot reshape array of size 1 into shape (144,256)
```

Any solutions?
TIA.

msb336, 2019-03-29:
The airsim server sometimes returns an empty image message, which is why reshaping causes this value error. We are still looking into why the image return is sometimes empty, but as a workaround, you can add a logical catch for when these empty images are returned:

```
if img1d.size == responses[0].height * responses[0].width:
    # reshape image
else:
    #image is empty
```

closed 2019-10-08 by msb336

rallen10, 2020-07-22:
I'm still getting this same error and it doesn't seem to just be me (https://github.com/microsoft/AirSim/issues/2169). Perhaps this issue should be opened again