microsoft/AirSim #1843: Confusing Vector3r behavior for multiplication and division for python

wfus, 2019-03-18:
Is there a reason why broadcasted division for `Vector3r` objects broadcasts for the `x, y, z` dimension while broadcasted multiplication for `Vector3r` only broadcasts for the `x, y` dimension and leaves the `z` dimension alone? Is this intended? This can be found in `PythonClient/airsim/types.py`.

```python
    def __truediv__(self, other):
        if type(other) in [int, float] + np.sctypes['int'] + np.sctypes['uint'] + np.sctypes['float']:
            return Vector3r( self.x_val / other, self.y_val / other, self.z_val / other)
        else: 
            raise TypeError('unsupported operand type(s) for /: %s and %s' % ( str(type(self)), str(type(other))) )

    def __mul__(self, other):
        if type(other) in [int, float] + np.sctypes['int'] + np.sctypes['uint'] + np.sctypes['float']:
            return Vector3r(self.x_val*other, self.y_val*other, self.z_val)
        else: 
raise TypeError('unsupported operand type(s) for *: %s and %s' % ( str(type(self)), str(type(other))) )
```
Have also seen this in my own scripts.

```python
import airsim
    def __truediv__(self, other):
        if type(other) in [int, float] + np.sctypes['int'] + np.sctypes['uint'] + np.sctypes['float']:
            return Vector3r( self.x_val / other, self.y_val / other, self.z_val / other)
        else: 
            raise TypeError('unsupported operand type(s) for /: %s and %s' % ( str(type(self)), str(type(other))) )

    def __mul__(self, other):
        if type(other) in [int, float] + np.sctypes['int'] + np.sctypes['uint'] + np.sctypes['float']:
            return Vector3r(self.x_val*other, self.y_val*other, self.z_val)
        else: 
raise TypeError('unsupported operand type(s) for *: %s and %s' % ( str(type(self)), str(type(other))) )
```

msb336, 2019-03-20:
@wfus Good find. I think this must be an error that went unnoticed until now.

wfus, 2019-03-20:
I'm assuming broadcasting to all 3 dimensions is intended? Might submit a PR later today, since I've changed the vector3r class in my own code.

closed 2019-03-29 by msb336