microsoft/AirSim #1844: Clarification in Logitech G920 Steering Wheel Installation

hrishikeshtawade04, 2019-03-18:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------
In the 4th step of the Logitech G920 Steering Wheel Installation it says to debug the Airsim Project. Does it mean we should rebuild it? If no how to debug it? 
Also it says further to go to Unreal Editor. In this case which project needs to be opened in Unreal editor to perform steps ahead? When I am choosing the blocks environment for editing it says that the 4.18 version is old and the environment is built in a newer version.

msb336, 2019-03-20:
@hrishikeshtawade04 to "debug the AirSim Project," open the Visual Studio project file (.vcxproj) of the unreal environment you want to use, and run the project through Visual Studio Debug mode.

There should be an option to rebuild your plugin libraries from source at the bottom of the message you are seeing about the version being outdated. Go ahead and click that option, and things should build correctly.

hrishikeshtawade04, 2019-03-22:
I am planning to use the Blocks environment. I went to Unreal/Environment/Blocks folder but couldn't find .vcxproj file. But there seems to be a .sln file. Can I use that?

hrishikeshtawade04, 2019-03-22:
@msb336 I am planning to use the Blocks environment. I went to Unreal/Environment/Blocks folder but couldn't find .vcxproj file. But there seems to be a .sln file. Can I use that?

hrishikeshtawade04, 2019-03-22:
@msb336 It worked with the .sln file too. I want to try it on the other available environments. Where can I get there .sln files?

msb336, 2019-04-10:
@hrishikeshtawade04 Sorry, it was the solution file (*.sln) not project file that you need. Unfortunately you cannot debug the other environments, as only their binaries are available to the community.

jonyMarino, 2020-10-19:
Closed due to inactivity. Doc update pending

closed 2020-10-19 by jonyMarino