microsoft/AirSim #1845: Image sampling interval is not fixed

JWDu, 2019-03-19:
Hi，everyone，when I collect images using REC button or the python API，the time interval is not fixed from the timestamp of the photo. For my computer, each image will cost about 95 ms to 105 ms, can I make the sampling rate stable such as fixed 100 ms?

tiax615, 2019-04-25:
Maybe you can decrease clockspeed to cover this issue?
for example, set clockspeed=0.1 in Setting.json. It looks like better.

jaymefosa, 2019-04-29:
Setting clockspeed = .1 for me throws a Json error "unexpected ."

jaymefosa, 2019-04-29:
Ah this needs to be 0.1, but it only affects the physics engine apparently, not animations for propellers or other things.

ironclownfish, 2020-05-19:
This should be done with a timer client-side. Make sure you leave enough time for each capture that your computer can capture each frame that quickly given the level performance.

closed 2020-05-19 by ironclownfish