microsoft/AirSim #1852: Incorrect Pallet at docs/seg_rgbs.txt (Windows + Unity + AirSim + CarDemo)

edessa, 2019-03-21:
Hello,

I've been using **Unity** and Airsim (on Windows, with the UnityDemo environment, and CarDemo chosen), but am running into an issue where the pallet in Airsim/docs/seg_rgbs.txt is incorrect, and so I have no way of knowing the mapping from ID to RGB. When I run the following code:
```
response_seg = client.simGetImage(0, airsim.ImageType.Segmentation)
np_seg = np.asarray(bytearray(response_seg))
dec_seg = cv2.imdecode(np_seg, cv2.IMREAD_COLOR)
palette = dec_seg .reshape(dec_seg .shape[0]*dec_seg .shape[1], dec_seg .shape[2])
palette_colors = np.unique(palette, axis=0)
print(palette_colors)
```
The following colors are printed:
```
[[127 127 127], [130   0   6], [130   8  22], [131   8  22], [131   8 162], [131  65  48], [131  72  32]
 ...]
```
and none of them are contained inside the AirSim/docs/seg_rgbs.txt file that contains the ID and Pallet mapping. Attached are the corresponding images (RGB and Segmentation):
[image] [image]

I saw a file in AirSim/Unity/UnityDemo/Assets/AirSimAssets/Scripts/HUD/ColorEncoding.cs , and the method of interest is EncodeIDAsColor, but that function isn't creating the outputs that I'm seeing above. Does anybody know the pallet for object IDs in Unity? All previous questions seems to be directed at AirSim. Attached is my [settings.json](https://github.com/Microsoft/AirSim/files/2994076/settings.txt) file.

I wouldn't need this mapping if I had access to a list of pixel-indices corresponding to each ID, but it doesn't seem AirSim provides that access. I'd be grateful to whoever is able to help, as I haven't been able to make any progress today because of this issue.

edessa, 2019-03-21:
Unless anyone comments, it looks like my only option is to iteratively change the IDs of an object in the scene and record the color. I can only hope the ID-color mapping is deterministic...

edessa, 2019-03-22:
I fixed the problems mentioned above through a couple hacks. The code in CameraFiltersScript.cs uses GameObject IDs which are not actually scaled from 0 - 255 at initialization. Furthermore, the SetSegmentationID  method in CameraFiltersScript.cs does not properly assign multiple IDs (with the regexing option), and only assigns a single ID at a time. Also, make sure to comment out the finalID bitshifting operation in Vehicle.cs (under the SetSegmentationID method), because it never gets un-set. You'll have to look at the EncodeIDAsColor method in ColorEncoding.cs as well and make it work for your purposes. This is pretty much all I had to do to get the segmentations working.

closed 2019-03-22 by edessa

averyehorvath, 2020-08-14:
I'm using Unreal but I pulled in the latest changes of Airsim and updated to Unreal 4.24 and that fixed it for me