microsoft/AirSim #1855: NED location and GPS

KeyneZzz, 2019-03-25:


DavidLSmyth, 2019-03-25:
If the distance is far enough the curvature of the earth may be responsible (equations that calculate the distance between gps coordinates takes the curvature of the earth into account). Do you have the figures outlining the inconsistency? Apologies if you've already ruled this out

KeyneZzz, 2019-03-26:
> If the distance is far enough the curvature of the earth may be responsible (equations that calculate the distance between gps coordinates takes the curvature of the earth into account). Do you have the figures outlining the inconsistency? Apologies if you've already ruled this out

@DavidLSmyth thanks for your reply, and I am not considering the curvature thing.
My route is firstly heading north for 1.5 kilometer, then then turn to east for 1.3 km. When arrived, it's about 20 meters northeast from the set endpoint. The latitude is about 30 degrees.

DavidLSmyth, 2019-03-26:
I did some of the calculations and as far as I could see there should be a small error in the gps location due to the non-planar geometry of the earth, the points are close enough to be considered on a plane. I'm not sure how the NED figures are calculated, it might be worth having a look at the source code. The calculation of GPS location is easy to find in the source code but it could be worth checking that too

madratman, 2019-03-29:
So, I did some digging:

the NED<->geodetic conversions are handled by `EarthUtils::nedToGeodetic` which you can inspect in https://github.com/Microsoft/AirSim/blob/123bbd6d6416bbe98e775101ba8ebcb5ccbf8501/AirLib/include/common/EarthUtils.hpp#L288-L304

Now, [PawnSimApi.cpp](https://github.com/Microsoft/AirSim/blob/39fb3b0b051050f4bc9d770838808756b64ff5ab/Unreal/Plugins/AirSim/Source/PawnSimApi.cpp) uses the above `EarthUtils::nedToGeodetic` in two places:
 -  to set the correspondence b/w home geo point (from settings.json) and unreal pawn's spawn coordinates and the NED frame. This can be seen in https://github.com/Microsoft/AirSim/blob/39fb3b0b051050f4bc9d770838808756b64ff5ab/Unreal/Plugins/AirSim/Source/PawnSimApi.cpp#L67
  
- to update the state of the pawn, as can be seen in https://github.com/Microsoft/AirSim/blob/39fb3b0b051050f4bc9d770838808756b64ff5ab/Unreal/Plugins/AirSim/Source/PawnSimApi.cpp#L76 you can see `PawnSimApi::update()` is being called in every tick, 
which is calling `environment_->update()` in https://github.com/Microsoft/AirSim/blob/39fb3b0b051050f4bc9d770838808756b64ff5ab/Unreal/Plugins/AirSim/Source/PawnSimApi.cpp#L305, which is calling `EarthUtils::nedToGeodetic` in the `msr::airlib::environment::updateState()` function as can be seen in 
https://github.com/Microsoft/AirSim/blob/123bbd6d6416bbe98e775101ba8ebcb5ccbf8501/AirLib/include/physics/Environment.hpp#L96

Now, [EarthUtils.hpp#L306](https://github.com/Microsoft/AirSim/blob/123bbd6d6416bbe98e775101ba8ebcb5ccbf8501/AirLib/include/common/EarthUtils.hpp#L306) does mention `"  //below are approximate versions and would produce errors of more than 10m for points farther than 1km
//for more accurate versions, please use the version in EarthUtils::nedToGeodetic"`, but they are for `GeodeticToNedFast` and `nedToGeodeticFast` (not the non-fast versions :)), which are not being used anywhere except in [droneshell here](https://github.com/Microsoft/AirSim/blob/123bbd6d6416bbe98e775101ba8ebcb5ccbf8501/DroneShell/src/main.cpp#L1190)
https://github.com/Microsoft/AirSim/blob/123bbd6d6416bbe98e775101ba8ebcb5ccbf8501/DroneShell/src/main.cpp#L1190
While, I haven't verified if `GeodeticConverter::nedToGeodetic`, we also have a function called `ned2Geodetic` in an unused header file in https://github.com/Microsoft/AirSim/blob/123bbd6d6416bbe98e775101ba8ebcb5ccbf8501/AirLib/include/common/GeodeticConverter.hpp#L128-L139
That might be worth a shot. So you can replace the calls of `EarthUtils::nedToGeodetic` in all the shebang I mention earlier with `GeodeticConverter::nedToGeodetic` and test that (or well, do the math heh)

Also, `GeodeticConverter.hpp` itself is based on another open source repo by ETH Zurich's Autonomous Systems Lab as can be seen in [ethz-asl/geodetic_utils](https://github.com/ethz-asl/geodetic_utils/blob/master/include/geodetic_utils/geodetic_conv.hpp#L142)
An interesting thing to watch out is issue https://github.com/ethz-asl/geodetic_utils/issues/28 , which seems to be accounted for in airlib in the following https://github.com/Microsoft/AirSim/blob/123bbd6d6416bbe98e775101ba8ebcb5ccbf8501/AirLib/include/common/GeodeticConverter.hpp#L136