microsoft/AirSim #1856: "simSetVehiclePose" Simulation Mode Change

Endf22, 2019-03-27:
Hello,

I want to use the "simSetVehiclePose" to respawn the drone randomly in the environment. However, whenever I use this function "simSetVehiclePose", the computer vision mode is activated and all physics of the drone are gone. Is it possible to switch back to "Multirotor" Mode using python API?

Thank you

-------------------------------
Ubuntu 18.04

AIRSIM settings.json
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor"

msb336, 2019-04-12:
@Endf22 are you using unity, or unreal engine?

Endf22, 2019-04-13:
Thank you for commenting. I am using Unreal Engine.

wuming0624, 2019-04-30:
Having the same issue here with Unity. Checked in #1113 #1272 still couldn't figure out. Any updates?

Endthere, 2019-05-01:
I still have the same problem unfortunately.

PR #3363 (2021-02-02): Fix losing physics after setting pose

Fixes: #1272
Fixes: #1856
Fixes: #1246
Fixes: #3049

## About
This is a slightly modified version of @songxiaocheng's PR originally at #2324. This PR includes all his changes with the exception of 1 line that still has an open question.

## How Has This Been Tested?
Build was confirmed to work, and a modified version of the teleport.py was tested where a moveToPositionAsync() API call was made after a simSetVehiclePose() call.

## Screenshots (if appropriate):