microsoft/AirSim #1857: ComputerVision mode with Unity

fwarmuth, 2019-03-28:
Hey,
I just look into AirSim with Unity. At the moment i am looking into generating segmentation images.
I'm try to get the ComputerVision mode running with Unity.
I'm using Windows 10 + VSC2017(15.9.9) + Unity(2018.2.15.f1)

Everything is freshly build andmy `settings.json` looks like:

```
{
  "SettingsVersion": 1.2, 
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md", 
  "SimMode": "ComputerVision" 
}
```

It ends up just crashing.
Is the `ComputerVision` mode working with Unity or anything else i need to check?
I looked through the docs but could not find anything or i am blind...

any help is appreciated.
thanks

howard-mahe, 2019-03-28:
Same here as well

msb336, 2019-03-29:
Unfortunately CV mode is not yet supported in Unity. We're working to get our Unity side up to speed with what airsim can do in Unreal, but there's still a long way to go!

howard-mahe, 2019-04-01:
Thank you msb336 for your feedback and for your work. I am already glad Airsim works on a scene such as Windridge City.

For bypassing the lack of `ComputerVision` mode in Unity, I am now used to run Airsim in `Drone` mode and set the position of the vehicle+camera using
```python
client.simSetVehiclePose(
   airsim.Pose(airsim.Vector3r(0, 0, 0), airsim.to_quaternion(0, 0, 0),
   False,
   ""
)
```

fwarmuth, 2019-04-03:
@msb336 thank you for your reply. Not what i was wishing for, but i guess somebody has to do the work - i am got no time to tackle that right now...

Hey howard,
I decided to just use the things i need out of AirSim. In my use case i just want to generate image segmentation and labeled data. Therefore i just use the "CameraEffects.shader" and the "ColorEncoding.cs" files. I wrote a exportation function and now going from there. But it seems not to be the optimal path. A lot of rewriting code other can write way better than me.
If i got time, or stuck, i will check out using the DroneMode.

clogsi, 2019-06-27:
Hey fwarmuth,
would you share the way, how you got the shader and ColorEncoding-class into Unity?

fwarmuth, 2019-07-03:
> 
> 
> Hey fwarmuth,
> would you share the way, how you got the shader and ColorEncoding-class into Unity?

Hey,
I remember no big problems. Just import (drag and drop) them to you unity project. 
If you got any problems post a error message.
regards

jackgale, 2020-03-05:
Hi @msb336 I'm curious if there are any updates, are we still unable to get AirSims scene segmentation + computer vision mode while using Unity?

nielspost, 2020-03-06:
> Thank you msb336 for your feedback and for your work. I am already glad Airsim works on a scene such as Windridge City thanks to the Unity plugin.
> 
> For bypassing the lack of `ComputerVision` mode in Unity, I am now used to run Airsim in `Drone` mode and set the position of the vehicle+camera using
> 
> ```python
> client.simSetVehiclePose(
>    airsim.Pose(airsim.Vector3r(0, 0, 0), airsim.to_quaternion(0, 0, 0),
>    False,
>    ""
> )
> ```
> 
> which disables the physics of the drone.

Interesting solution, would like to implement this.
Do you have more information on which file and lines I edit with this code?

Thanks in advance